allangood/rtlamr2mqtt #1: Extend for multiple items in FILTER_ID

sradner13, 2021-05-08:
Thank you for publishing this.  Works great for a single meter, but in the no good deed goes unpunished category; any interest in working with multiple meters?

Would mean changing the mqtt publish to use fld 3 instead of the static name.

Probably not your use case, so feel free to close and tell me to go away.

Cheers and thanks.

allangood, 2021-05-08:
Hello @sradner13 , I really wasn't expecting anyone using this! :D
I did a quick thing for my very specific need that is the water meter used in Winnipeg and that is the reason this repository doesn't even have a README file. Now I have a reason to make it better! :)

Do you mind to send an RTLAMR output for your meters so I can think in a way to make the script more flexible for different needs?
Also, if you could share the protocol used by your meters (SCM, SCM+, etc) would be very useful!

Thank you!

allangood, 2021-05-08:
I was wondering if a configuration file like this this one would make sense for you:

```
mqtt:
  host: 192.168.1.1
  user: mqtt
  password: sdfhkjh(*&

meters:
  - id: 7823010
    protocol: scm+
    name: meter_water
    id_field: 3
    read_field: 7
  - id: 6567984
    protocol: scm
    name: meter_hydro
    id_field: 2
    read_field: 3
```
Do you think it would be difficult to get these information?

sradner13, 2021-05-09:
I was using a Pi3 to run something very similar, so when I saw you had packaged for Docker it was the perfect solution.

I see hundreds of meters in my neighborhood.  Here are three examples.  SCM is gas or electric / R900 is a water meter.

{Time:2021-05-08T23:36:26.484 SCM:{ID:39983207 Type: 4 Tamper:{Phy:00 Enc:01} Consumption:   65035 CRC:0xCAA5}}
{Time:2021-05-08T23:36:26.646 R900:{ID:1551538194 Unkn1:0xA3 NoUse:34 BackFlow:0 Consumption: 1322625 Unkn3:0x00 Leak: 0 LeakNow:0}}
{Time:2021-05-09T00:04:35.289 SCM+:{ProtocolID:0x1E EndpointType:0x6E EndpointID: 101406523 Consumption:  91467859 Tamper:0x4089 PacketCRC:0xC3B8}}

In CSV form data is normalized for scm & r900, so in my examples the id is in field 3 and the consumption is field 7.  The scm+ example I grabbed seems to move the id field, but the consumption is still field 7.

2021-05-08T23:31:07.594981803Z,0,0,39983207,4,0x0,0x1,67743,0xa9df
2021-05-08T23:31:07.650158882Z,0,0,1551538194,12,0x0,0x0,653696,0x67ad
2021-05-09T00:04:13.392579852Z,0,0,0x16a3,0x1e,0x6e,101406523,90603393,0x4089,0x5089

Currently I/m using environment variables.
name MQTT_HOST /  value   mqtt.domain.com
name PROTOCOL /  value     scm,r900
name FILTER_ID /  value        39984077,1831274292,61145418

You config solution would work fine.  I did note you added a decimal point in your script, not sure if you would want to make that configurable as well.

Thanks for looking at all this.

allangood, 2021-05-09:
Yes, that is true. My water meter uses a decimal point.
Adding an optional "format" option may help with this.
Another think that I forgot is the unit of measurement (kWh, m³, etc)
```
mqtt:
  host: 192.168.1.1
  user: mqtt
  password: sdfhkjh(*&

meters:
  - id: 7823010
    protocol: scm+
    name: meter_water
    id_field: 3
    read_field: 7
    format: XXXXX.XXX
    unity: m³
  - id: 6567984
    protocol: scm
    name: meter_hydro
    id_field: 2
    read_field: 3
    unit: kWh
```

A non related question: Does your Pi run really hot after you started to use RTLAMR? Mine went from average 55 °C/133 °F to 72 °C/161 °F.

That was the reason I've made a change to kill RTL_TCP and RTLAMR after the readings. This helped to keep my Pi cool.

I am asking this to know if I should consider doing the same thing with multiple meters.

Thank you.

sradner13, 2021-05-09:
The RTL-SDR dongle itself is always very warm, this seems to be normal.  The Pi was also running hot.  rtlamr seems to consume 100% of the cpu at all times.  I switched the Pi to POE with a fan, and the fan was running almost constantly.  The Docker install is on a full fledge AMD Ryzen server, and rtlamr still likes to consume most of a core. I've got cores to spare in the server, so I haven't investigated the rtlamr code to see why it's a CPU hog.

My problem is my gas meter runs in wake-up mode instead of bubble-up mode.  I have hacked together a really low power, battery powered transmitter which wakes it up every two hours, but I can't really control the exact timing, so I need to listen at all times for incoming data.  At some point I might rebuild the wake-up transmitter to run on AC and then it could coordinate wake up time with active time on rtlamr, but have not done that project yet.  Maybe a config for sleep / continuous mode?

allangood, 2021-05-10:
Just pushed my changes and the container will be ready soon!

allangood, 2021-05-10:
I am very glad to hear back from you!

Let's see:

1. You can still use volumes to mount a single file. The README was updated to show how to do so:
```
volumes:
      /etc/rtlamr2mqtt.yaml:/etc/rtlamr2mqtt.yaml:ro
```
2. The README has the most up to dated information. All parameters there are the correct ones. The config sample used in this issue history was just a brain storm :)
3.  Yes, this is on my TODO list. Would be useful to have a section in the config file to add custom RTLTCP/RTLAMR configuration. One of the parameters accepted by RTLAMR is to remove duplicate messages (-unique=true). It may be a good idea to add this by default
4. I am using the format and it is working very well. Do you mind to share your redacted config file?
5. Hahaha! No need for that, my I really appreciate! I do it for fun! And I like the feedback :)

sradner13, 2021-05-10:
On the volume mount, I manage docker using portainer rather than command line.  It will only mount volumes, won't do a file, I can bind a file, so no worries.

Here's my config:
mqtt:
  host: mqtt.domain.com
  password:
  ha_autodiscovery: true
  ha_autodiscovery_topic: homeassistant

meters:
  - id: 39984077
    protocol: scm
    name: electric
    field_meterid: 3
    field_consumption: 7
    format: #####.###
    unit_of_measurement: kWh
  - id: 61145418
    protocol: scm
    name: gas
    field_meterid: 3
    field_consumption: 7
    unit_of_measurement: cuFt
  - id: 1831274292
    protocol: r900
    name: water
    field_meterid: 3
    field_consumption: 7
    uniti_of_measurement: cuFt

I'm running without the format line and getting all good stuff.

allangood, 2021-05-10:
Ah, I see. I forgot to add this in the README, but you have to put the format parameter inside quotes. Like this:
```
id: 39984077
protocol: scm
name: electric
field_meterid: 3
field_consumption: 7
format: "#####.###"
unit_of_measurement: kWh
```

Otherwise it is considered a comment.
I am updating the README!

sradner13, 2021-05-10:
custom_parameter did the trick.  Getting all the messages I would expect, and OpenHAB even picked up the homeassistant discovery information.  Hopefully others will benefit from your work.

One last thing, In your readme.MD you have this block.
  - id: 6567984
    protocol: scm
    name: meter_hydro
    **id_field**: 3
    **read_field**: 7
    **unit:** kWh

I think the bold fields were your working names...

allangood, 2021-05-10:
Ah! Good catch! :)
I totally missed these!

Thank you very much for all your great feedback!

I am closing the issue now as everything seems to be addressed!

Feel free to reopen or open new issues if needed!

Regards.

closed 2021-05-10 by allangood