allangood/rtlamr2mqtt #3: docker container never actually starts, stuck in restarting

tehbra1n, 2021-05-12:
After starting the container, it never actually starts:

```
CONTAINER ID   IMAGE                           COMMAND                  CREATED         STATUS                             PORTS                      NAMES
fa842b59ba88   allangood/rtlamr2mqtt           "/usr/bin/rtlamr2mqt…"   5 minutes ago   Restarting (1) 1 second ago                                   rtlamr2mqtt

```

dmesg shows

```
[   25.414327] br-9b530f50f209: port 1(vethd7fc5fc) entered blocking state
[   25.414335] br-9b530f50f209: port 1(vethd7fc5fc) entered disabled state
[   25.416589] device vethd7fc5fc entered promiscuous mode
[   25.424634] br-9b530f50f209: port 1(vethd7fc5fc) entered blocking state
[   25.424640] br-9b530f50f209: port 1(vethd7fc5fc) entered forwarding state
[   25.424692] IPv6: ADDRCONF(NETDEV_CHANGE): br-9b530f50f209: link becomes ready
[   25.434009] br-9b530f50f209: port 1(vethd7fc5fc) entered disabled state
[   25.435772] br-7412938b5455: port 1(vethe665982) entered blocking state
[   25.435777] br-7412938b5455: port 1(vethe665982) entered disabled state
[   25.435925] device vethe665982 entered promiscuous mode
[   25.436137] br-7412938b5455: port 1(vethe665982) entered blocking state
[   25.436141] br-7412938b5455: port 1(vethe665982) entered forwarding state
[   25.445398] br-7412938b5455: port 1(vethe665982) entered disabled state
[   26.091322] cgroup: cgroup: disabling cgroup2 socket matching due to net_prio or net_cls activation
[   26.879720] eth0: renamed from veth9c9fd90
[   26.893545] IPv6: ADDRCONF(NETDEV_CHANGE): vethe665982: link becomes ready
[   26.893623] br-7412938b5455: port 1(vethe665982) entered blocking state
[   26.893628] br-7412938b5455: port 1(vethe665982) entered forwarding state
[   26.893722] IPv6: ADDRCONF(NETDEV_CHANGE): br-7412938b5455: link becomes ready
[   26.960433] eth0: renamed from veth5d407f4
[   26.980259] IPv6: ADDRCONF(NETDEV_CHANGE): vethd7fc5fc: link becomes ready
[   26.980348] br-9b530f50f209: port 1(vethd7fc5fc) entered blocking state
[   26.980352] br-9b530f50f209: port 1(vethd7fc5fc) entered forwarding state
[   27.907908] br-9b530f50f209: port 1(vethd7fc5fc) entered disabled state
[   27.908199] veth5d407f4: renamed from eth0
[   28.003419] br-9b530f50f209: port 1(vethd7fc5fc) entered disabled state
[   28.010869] device vethd7fc5fc left promiscuous mode
[   28.010884] br-9b530f50f209: port 1(vethd7fc5fc) entered disabled state
[   28.163194] br-9b530f50f209: port 1(veth74f3aa0) entered blocking state
[   28.163200] br-9b530f50f209: port 1(veth74f3aa0) entered disabled state
[   28.163388] device veth74f3aa0 entered promiscuous mode
[   28.163652] br-9b530f50f209: port 1(veth74f3aa0) entered blocking state
[   28.163656] br-9b530f50f209: port 1(veth74f3aa0) entered forwarding state
[   28.919834] br-9b530f50f209: port 1(veth74f3aa0) entered disabled state
[   29.500723] eth0: renamed from vethbe5c8c9
[   29.532076] IPv6: ADDRCONF(NETDEV_CHANGE): veth74f3aa0: link becomes ready
[   29.532169] br-9b530f50f209: port 1(veth74f3aa0) entered blocking state
[   29.532172] br-9b530f50f209: port 1(veth74f3aa0) entered forwarding state
[   30.260343] br-9b530f50f209: port 1(veth74f3aa0) entered disabled state
[   30.260492] vethbe5c8c9: renamed from eth0
[   30.336197] br-9b530f50f209: port 1(veth74f3aa0) entered disabled state
[   30.351150] device veth74f3aa0 left promiscuous mode
[   30.351166] br-9b530f50f209: port 1(veth74f3aa0) entered disabled state
[   30.501904] br-9b530f50f209: port 1(veth7761b7d) entered blocking state
[   30.501909] br-9b530f50f209: port 1(veth7761b7d) entered disabled state
[   30.502248] device veth7761b7d entered promiscuous mode
[   30.504985] br-9b530f50f209: port 1(veth7761b7d) entered blocking state
[   30.504992] br-9b530f50f209: port 1(veth7761b7d) entered forwarding state
[   30.935935] br-9b530f50f209: port 1(veth7761b7d) entered disabled state
[   31.627312] eth0: renamed from veth22f25ee
[   31.642134] IPv6: ADDRCONF(NETDEV_CHANGE): veth7761b7d: link becomes ready
[   31.642235] br-9b530f50f209: port 1(veth7761b7d) entered blocking state
[   31.642239] br-9b530f50f209: port 1(veth7761b7d) entered forwarding state
[   32.245137] veth22f25ee: renamed from eth0
[   32.315692] br-9b530f50f209: port 1(veth7761b7d) entered disabled state
[   32.353475] br-9b530f50f209: port 1(veth7761b7d) entered disabled state
[   32.360939] device veth7761b7d left promiscuous mode
[   32.360958] br-9b530f50f209: port 1(veth7761b7d) entered disabled state
[   32.635660] br-9b530f50f209: port 1(vethc75a80d) entered blocking state
[   32.635666] br-9b530f50f209: port 1(vethc75a80d) entered disabled state
[   32.635998] device vethc75a80d entered promiscuous mode
[   32.643163] br-9b530f50f209: port 1(vethc75a80d) entered blocking state
[   32.643169] br-9b530f50f209: port 1(vethc75a80d) entered forwarding state
[   32.966455] br-9b530f50f209: port 1(vethc75a80d) entered disabled state
```

I'm sure I'm missing something obvious here.  This is on an Ubuntu box:

```
Distributor ID: Ubuntu
Description:    Ubuntu 20.04.2 LTS
Release:        20.04
Codename:       focal
```

Using a Nooelec NESDR Mini.

allangood, 2021-05-13:
Hello @tehbra1n ,

This seems to be something related with the configuration file or the device.

Could you share the output of these 2 commands, please?
`lsusb`
`docker logs --tail=50 rtlamr2mqtt`

Also, the content of you configuration file:
`cat /etc/rtlamr2mqtt.conf`

Thank you!

tehbra1n, 2021-05-13:
OK, lsusb output:

```
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 8087:0ab6 Intel Corp. UDOO X86
Bus 001 Device 002: ID 0bda:2838 Realtek Semiconductor Corp. RTL2838 DVB-T
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

docker logs:
```

Traceback (most recent call last):
  File "/usr/bin/rtlamr2mqtt.py", line 34, in <module>
    with open('/etc/rtlamr2mqtt.yaml','r') as config_file:
IsADirectoryError: [Errno 21] Is a directory: '/etc/rtlamr2mqtt.yaml'
Traceback (most recent call last):
  File "/usr/bin/rtlamr2mqtt.py", line 34, in <module>
    with open('/etc/rtlamr2mqtt.yaml','r') as config_file:
IsADirectoryError: [Errno 21] Is a directory: '/etc/rtlamr2mqtt.yaml'
Traceback (most recent call last):
  File "/usr/bin/rtlamr2mqtt.py", line 34, in <module>
    with open('/etc/rtlamr2mqtt.yaml','r') as config_file:
IsADirectoryError: [Errno 21] Is a directory: '/etc/rtlamr2mqtt.yaml'
Traceback (most recent call last):
  File "/usr/bin/rtlamr2mqtt.py", line 34, in <module>
    with open('/etc/rtlamr2mqtt.yaml','r') as config_file:
IsADirectoryError: [Errno 21] Is a directory: '/etc/rtlamr2mqtt.yaml'
Traceback (most recent call last):
  File "/usr/bin/rtlamr2mqtt.py", line 34, in <module>
    with open('/etc/rtlamr2mqtt.yaml','r') as config_file:
IsADirectoryError: [Errno 21] Is a directory: '/etc/rtlamr2mqtt.yaml'
Traceback (most recent call last):
  File "/usr/bin/rtlamr2mqtt.py", line 34, in <module>
    with open('/etc/rtlamr2mqtt.yaml','r') as config_file:
IsADirectoryError: [Errno 21] Is a directory: '/etc/rtlamr2mqtt.yaml'
```

I don't recall creating a rtlamr2mqtt.yaml directory in /etc manaully.

rtlarm2mqtt.conf:

```
mqtt:
  host: 192.168.1.1
  user: mqtt
  password: sdfhkjh(*&
  ha_autodiscovery: true
  ha_autodiscovery_topic: homeassistant
```

I used just the mandatory part of the config file from the example while I get the radio part up and working.  I'll replace the example settings with those of my mqtt broker once the container works.

Thanks!

allangood, 2021-05-13:
The error log says the rtlarm2mqtt.conf is a directory.
This happens when you run the docker container without an **existent** file. I.e. you have to create the rtlarm2mqtt.conf before starting the container.

Do you have a /etc/rtlarm2mqtt.conf file?

tehbra1n, 2021-05-13:
Yes, the /etc/rtlarm2mqtt.conf file was included in the previous post.  I only included the mqtt section that was mandatory.

The error log refers to /etc/rtlarm2mqtt.**yaml** but the config file is called /etc/rtlarm2mqtt.**conf**.,  Should the config file extension be .yaml?

allangood, 2021-05-13:
Short answer: Yes, you have to rename your config file.

Long answer: The file name it really doesn't matter, as soon you point the mapping to the right file. 

Looks like you are using the docker-compose example from the main page, it has a section like this:
```
volumes:
      - /etc/rtlamr2mqtt.yaml:/etc/rtlamr2mqtt.yaml:ro
```
It is "source" : "destination". The "source" is the file in the host machine and the "destination" is the file inside the container.
As the source file doesn't exist, docker-compose creates a directory and map it to the container. That is what is happening with your machine.

tehbra1n, 2021-05-13:
I renamed the config file, but there were still errors.  It seems like the "meters" section is mandatory and it doesn't want to start without the mqtt broker being correct. So I got those going.

What is the simplest way to see what meters the radio is picking up?

allangood, 2021-05-13:
Oh, you are not trying to run the container, but use it to inspect the traffic. This is a whole different thing.

To do what you want, you have to run this command:
```
docker run --rm -ti --device /dev/bus/usb/001/002 allangood/rtlamr2mqtt /bin/ash
```
Then you will be inside the container.
From there you have to run:
```
rtl_tcp &
rtlamr -msgtype=all
```
This will output all meters in the range. To break the execution, press `crtl + c`

To quit the container, type `exit`

tehbra1n, 2021-05-14:
ok, the results then are:

```
$ docker run --rm -ti --device /dev/bus/usb/001/002 allangood/rtlamr2mqtt /bin/ash
/tmp # rtl_tcp &
/tmp # rtlamr -msgtype=allFound 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Tuned to 100000000 Hz.
listening...
Use the device argument 'rtl_tcp=127.0.0.1:1234' in OsmoSDR (gr-osmosdr) source
to receive samples in GRC and control rtl_tcp parameters (frequency, gain, ...).

```
and that's it.  

So if nothing appear after that after a while, I assume there are no meters in range?

allangood, 2021-05-14:
That is correct.

Here is a list of tested meters:
https://github.com/bemasher/rtlamr/blob/master/meters.csv

I am closing this as it is not an issue with the project but support request.

closed 2021-05-14 by allangood

tehbra1n, 2021-05-14:
Got it, thanks!