allangood/rtlamr2mqtt #4: Crashing after scm+ messages come in

mloebl, 2021-07-16:
I ran this for months without any problems, and starting today after running for a short period whenever an scm+ message comes in, it looks like rtlamr is crashing as see:
```
comm recv bye
worker cond timeout
all threads dead..
listening...
Use the device argument 'rtl_tcp=127.0.0.1:1234' in OsmoSDR (gr-osmosdr) source
to receive samples in GRC and control rtl_tcp parameters (frequency, gain, ...).
```
It never comes back. It runs very stably as well until the scm+ message comes in. Some history here too, my gas meter was replaced today with an scm+ model, hence trying this out today. I went into the container manually and ran the script in /usr/bin/rtlamr2mqtt.py , but can't see where it's crashing, but betting on the decode of the scm+ traffic as dies as soon as a messages comes in.

Thank you for the hard work!

allangood, 2021-07-16:
Hi, thank you for your time to report this error!

I know that is It not trivial to debug this container, but could you log into the container and execute these commands and send the output back, please?
assuming the container name is rtlamr2mqtt:
```
docker exec -ti rtlamr2mqtt /bin/ash
rtl_tcp &
rtlamr -msgtype=scm+
```
The first command will log you into the container and then the other 2 commands are the commands I am using to get the meter readings.

Thank you

mloebl, 2021-07-16:
Thank you for the quick response :)  Wondering if the scm+ is a red herring, it's running absolutely fine and up for a few mins so far with the scm+ messages or something with the python decoding? Does it log anywhere if it has an errors?

allangood, 2021-07-16:
The python code should output in the console/std error.
If you run the rtlamr2mqtt.py command it crash, it should output something.

Try to get the latest output from the container:
```
docker logs --tail=100 rtlamr2mqtt
```

mloebl, 2021-07-16:
```Sending MQTT autodiscovery payload to Home Assistant...
RTL_TCP started with PID 7
listening...
Use the device argument 'rtl_tcp=127.0.0.1:1234' in OsmoSDR (gr-osmosdr) source
to receive samples in GRC and control rtl_tcp parameters (frequency, gain, ...).
RTLAMR started with PID 12
client accepted!
set freq 912600155
set sample rate 2359296
set gain mode 0
Meter "42378543" - Consumption 6917855. Sending value to MQTT.
comm recv bye
worker cond timeout
all threads dead..
listening...
Use the device argument 'rtl_tcp=127.0.0.1:1234' in OsmoSDR (gr-osmosdr) source
to receive samples in GRC and control rtl_tcp parameters (frequency, gain, ...).
```
This time failed on a single SCM message, it's strange.

allangood, 2021-07-16:
And how your configuration file looks like?

allangood, 2021-07-16:
Fro the last output, I can't see the Python code crashing, but the rtl_tcp was started twice. Probably it was the one that crashed.
The python code keep looking for the process and when it is not running, it will try to start it again. But it shouldn't crash in the first place :)

mloebl, 2021-07-16:
```
# (Optional section)
general:
  # Sleep for this amount of seconds after one successful of every meter
  # Set this to 0 (default) to disable it
  sleep_for: 300

# (Required section)
mqtt:
  host: REMOVED
  user: rtlamr2mqtt
  port: 1883
  password: REMOVED
  ha_autodiscovery: true
  ha_autodiscovery_topic: homeassistant

# (Optional)
custom_parameters:
  rtltcp: "-s 2048000"
  # ***DO NOT ADD -msgtype, -filterid nor -protocol parameters here***
  rtlamr: "-unique true -symbollength 7"

meters:
  #  - id: 75990052
  #  protocol: scm+
  #  name: meter_natgrid_gas
  #  format: "#####.###"
  #  unit_of_measurement: "CCF"
  #  icon: mdi:gauge
  - id: 42378543
    protocol: scm
    name: meter_natgrid_electric
    unit_of_measurement: kWh
```

haha that's what I thought :)  I'm trying to figure out what's impacting it... the only other thing I can think of is I moved to a new mqtt server since I used this container last. However I do see messages show up there before it crashes.  Wonder if I should try another system, I know docker isolates the container, but wondering if something else going on with that system.

allangood, 2021-07-16:
The Python code terminates all process after a successful reading to save memory if "sleep_for" > 0

I am not seeing any indication for a "crash" in the output, only the normal termination after a successful message.

Could you try to run the container again with "sleep_for" = 0 and see if it is what do you expect?

mloebl, 2021-07-16:
So far, so good with it set to 0. :) I'll let it run for a bit. Thank you very much for helping on a Friday night.

allangood, 2021-07-16:
The sleep_for > 0 will put the code to "sleep" and, at least for me, it doesn't make sense to keep all subprocess running while the code is just sleeping. That is why I've decided to kill all subprocess if is set to sleep.

Maybe a better documentation would have helped?

mloebl, 2021-07-16:
What was interesting is that what I was seeing was 1, 2 or sometimes 5 messages would come in, and wasn't seeing it sleep, or as I expected it to sleep for 300s when I watched standard out. Having it feed real time is absolutely fine in retrospect. So if I had just waited 5mins, I would have likely been fine. Well that was embarrassing. Again thank you very much for the help. It's safe to close this I think.

allangood, 2021-07-16:
I am closing this for now.
Feel free to reopen or open a new one if you find something. :)

Thank you for your feedback!

closed 2021-07-16 by allangood