allangood/rtlamr2mqtt #6: Home Assistant Add-On

AnthonyPluth, 2021-09-07:
Would you open to making your project installable as a Home Assistant Add-On? The amount of work to do so is pretty minimal and if I can get a better understanding of your use-case, I think I can put this together in a way that works with your existing workflow.

allangood, 2021-09-08:
Hello @AnthonyPluth ,

Thank you for your contributions.

From the feedback I have received so far, many users uses this project as a standalone container in some remote machine, like myself.

I have my HA instance running in a NUC in my Living Room and I have an Orange Pi running this docker container in my basement. That Pi in the basement doesn't have HA. How this would work been a HA Add-On in a machine without HA?

Unless I create some sort of API to make it possible to configure it remotely what would open a new world of work and security problems that I don't want to deal with...

AnthonyPluth, 2021-09-08:
@allangood It should be possible to use the same docker image for both a standalone service and an Add-On. By doing this, there shouldn't be any change in maintainability

allangood, 2021-09-09:
I don't have experience creating HA Add-ons.
The previous PR had changes to remove YAML what makes configuration much difficult.
How does it will work?

AnthonyPluth, 2021-09-09:
Yeah sorry about that, that was a bit aggressive. Now that I understand your use case, I think a better solution would be to add the option to configure via json with the default being yaml

allangood, 2021-09-10:
Let me create a new branch and we can work on this together, what do you thing?

AnthonyPluth, 2021-09-10:
let's do it!

allangood, 2021-09-13:
I have been fixing a few problems reported in the last days, the project has changed a bit since the last PR.

A new branch `ha-addon` has been created.

allangood, 2021-09-16:
Merged your PR and merged main with ha-addon branch.
Now I am fixing some issues to make Travis happy again!

Thank you for your contribution!

allangood, 2021-09-16:
Travis is happy again.
Now I will be working to fix the github action to push all branches to a tag on GitHub!

Let's see how it goes!

allangood, 2021-09-23:
Finally I've managed to fix my Github action to build multi-arch containers using only new actions. All deprecated action have been removed.
Merges to the ha-addon branch will trigger a new build/push to Docker.

allangood, 2021-09-24:
@AnthonyPluth , the first image has been pushed to Dockerhub and it is ready to be tested!

https://hub.docker.com/r/allangood/rtlamr2mqtt/tags?page=1&name=ha-addon

Thank you very much for all your effort! :)

allangood, 2021-09-24:
@AnthonyPluth ,I don't have or use HASSIO, could test the image, please?

AnthonyPluth, 2021-09-29:
Sorry for the delay - this does work, we just need to update the json config schema to match the updated yaml config. I'll try to get a PR in for this tomorrow.

One more thing - you don't need to build an image for the addon; this gets built locally when installed

datdamnzotz, 2021-10-05:
Works just fine - 
Took the directory in this branch https://github.com/allangood/rtlamr2mqtt/tree/ha-addon/rtlamr2mqtt-addon and dropped it in add-ons.
Ran the Supervisor -> Addons -> Reload.
Installed it.

I had to update config.json to (was complaining about normal,test) loaded it up and away she went.
```{
    "name": "rtlamr2mqtt",
    "version": "0.10.3",
    "slug": "rtlamr2mqtt",
    "panel_icon": "mdi:gauge",
    "description": "RTLAMR to MQTT Bridge",
    "startup": "application",
    "boot": "auto",
    "map": [
        "config:rw"
    ],
    "url": "https://gitlab.com/abpluth/rtlamr2mqtt",
    "uart": true,
    "udev": true,
    "usb": true,
    "host_network": false,
    "arch": [
        "amd64",
        "armv7",
        "i386"
    ],
    "services": ["mqtt:need"],
    "options": {
        "general": {
            "sleep_for": 300,
            "mode": "normal",
            "debug": false
        },
        "mqtt": {
            "host": "hassio.local",
            "port": 1883,
            "user": "",
            "password": "",
            "ha_autodiscovery": true,
            "ha_autodiscovery_topic": "homeassistant"
        },
        "rtl": {
            "rtltcp": "-s 2048000",
            "rtlamr": "-unique=true"
        },
        "meters": [
            {
                "id": 123456789,
                "protocol": "scm",
                "name": "gas_meter",
                "format": "####.###",
                "unit_of_measurement": "m³",
                "icon": "mdi:gas_canister"
            }
        ]
    },
    "schema": {
        "general": {
            "sleep_for": "int",
            "mode": "str",
            "debug": "bool?"
        },
        "mqtt": {
            "host": "str?",
            "port": "int?",
            "user": "str?",
            "password": "str?",
            "ha_autodiscovery": "bool?",
            "ha_autodiscovery_topic": "str?"
        },
        "rtl": {
            "rtltcp": "str?",
            "rtlamr": "str?"
        },
        "meters": [
            {
                "id": "int",
                "protocol": "list(idm|netidm|r900|r900bcd|scm|scm+)",
                "name": "str",
                "format": "str?",
                "unit_of_measurement": "str",
                "icon": "str"
            }
        ]
    }
}
```

I assume we are terminating and just relaunching after 5 mins?

```Sending MQTT autodiscovery payload to Home Assistant...
RTL_TCP started with PID 8
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001
Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Tuned to 100000000 Hz.
listening...
RTLAMR started with PID 13
Allocating 15 zero-copy buffers
Meter "32907488" - Consumption 113810. Sending value to MQTT.
Sleep_for defined, time to sleep!
Terminating all subprocess...
Signal caught, exiting!
Signal caught, exiting!
Signal caught, exiting!
RTL_TCP terminated.
RTLAMR terminated.
Sleeping for 300 seconds, see you later...
RTL_TCP started with PID 24
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001
Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Tuned to 100000000 Hz.
listening...
RTLAMR started with PID 29
Allocating 15 zero-copy buffers
Meter "32907488" - Consumption 113812. Sending value to MQTT.
Sleep_for defined, time to sleep!
Terminating all subprocess...
Signal caught, exiting!
Signal caught, exiting!
Signal caught, exiting!
RTL_TCP terminated.
RTLAMR terminated.
Sleeping for 300 seconds, see you later...```

allangood, 2021-10-13:
Thank you very much @AnthonyPluth for your contribution!
I am closing this issue now. Feel free to re-open or open new issues!

Regards

closed 2021-10-13 by allangood

allangood, 2021-10-20:
@jeffeb3 you can check my first Dockerfile version here:
https://github.com/allangood/rtlamr2mqtt/blob/0a7332bd9fb394d03d471d36046d874766f82acb/Dockerfile

PR #24 (2021-09-19): Add Home Assistant Addon
closes #6