allangood/rtlamr2mqtt #9: Everything looks right, I'm seeing meters in debug, but I'm not getting anything sent to mqtt

ponchohoncho, 2021-09-09:
Hello,

I'm not sure what's misconfigured, but I'm not getting any meter or autodiscover data sent to my MQTT broker. This is the contents of my mqtt section

```
mqtt:
  host: 192.168.1.xx   #xx is the number of my hassio VM, running mosquitto
  user: mymqttuser
  #password: "mymqttpassword"
  ha_autodiscovery: true
  ha_autodiscovery_topic: homeassistant
```

I also put intentionally wrong config info in there, and I didn't get any messages in the logs, either, so I'm not sure is it's managing to connect at all or not.

ponchohoncho, 2021-09-09:
fwiw, this is what the docker logs for your container look like for me

```
Starting in DEBUG Mode...
listening...
{"Time":"2021-09-09T17:37:04.480287828Z","Offset":0,"Length":0,"Message":{"ID":xxxxx928,"Type":12,"TamperPhy":3,"TamperEnc":0,"Consumption":238360,"ChecksumVal":9190}}

{"Time":"2021-09-09T17:37:13.821391357Z","Offset":0,"Length":0,"Message":{"ID":yyyyy738,"Type":12,"TamperPhy":2,"TamperEnc":0,"Consumption":682008,"ChecksumVal":38380}}

{"Time":"2021-09-09T17:37:18.848854257Z","Offset":0,"Length":0,"Message":{"ID":xxxxx314,"Type":12,"TamperPhy":3,"TamperEnc":0,"Consumption":236312,"ChecksumVal":16941}}

...repeat json string forver
```

I see other `print` messages in your python script, but none of them show up in the docker logs, so I'm not sure what I'm missing

bachya, 2021-09-09:
Same here. I have a similar config:

```yaml
mqtt:
  host: 172.16.30.xx
  user: rtlamr
  password: REDACTED

meters:
  # ...
```

...and am seeing this in the logs:

```
Meter "xxxx" - Consumption 33792.054. Sending value to MQTT.
...
```

...but nothing shows up in my MQTT broker. I've verified manually that the IP/port, username, and password all work.

bachya, 2021-09-09:
Some more info. I'm monitoring two meters:

* Electricity: SCM+ (ID: `1234`)
* Gas: SCM (ID: `5678`)

I patched `rtlamr2mqtt.py` to output `paho-mqtt`'s DEBUG-logs and saw this:

```
Meter "51643176" - Consumption 0441.362. Sending value to MQTT.
DEBUG:root:Sending CONNECT (u1, p1, wr0, wq0, wf0, c1, k60) client_id=b'rtlamr2mqtt'
DEBUG:root:Sending PUBLISH (d0, q0, r1, m2), 'b'rtlamr/gas_meter/state'', ... (8 bytes)
DEBUG:root:Sending DISCONNECT

Meter "100583244" - Consumption 33792.124. Sending value to MQTT.
DEBUG:root:Sending CONNECT (u1, p1, wr0, wq0, wf0, c1, k60) client_id=b'rtlamr2mqtt'
DEBUG:root:Sending PUBLISH (d0, q0, r1, m3), 'b'rtlamr/electricity_meter/state'', ... (9 bytes)
DEBUG:root:Sending DISCONNECT
``

Only the electricity meters shows up in my MQTT broker:

![CleanShot 2021-09-09 at 11 58 25@2x](https://user-images.githubusercontent.com/47216/132738180-1c064478-adae-4209-8205-68795121ac33.png)

allangood, 2021-09-09:
Hello @bachya , thanks for your time on this!

Are you using mosquitto? Do you have ACLs in place?

bachya, 2021-09-09:
@allangood I'm using VerneMQ. It has an ACL enabled by default, but since one meter comes through, I can't imagine that's an issue?

allangood, 2021-09-09:
Hi @ponchohoncho , looks like you are running the container in DEBUG mode. In this mode nothing will be sent to MQTT, it will only output meter readings in the screen.
This mode was created to help people to find their meters IDs. From your output looks like it is working as intended.

If you want to run in NORMAL mode, then you have to remove the `-e DEBUG=true` from your Docker call.

allangood, 2021-09-09:
@bachya it really depends on how the ACL is configured. I don't know how VerneMQ handle the ACLs, but on Mosquitto is possible to set ACLs for specific topics like:
```
user rtlamruser
topic readwrite rtlamr/gas_meter/#
```

From your message is possible to see the container is trying to send 2 messages.

Could you double check the ACLs?

bachya, 2021-09-09:
@allangood Turning the ACL off makes that topic show up, which is good. That said, I don't want to just leave it disabled. I've never had to manually configure user/topic combos in the ACL; why would that start now?

allangood, 2021-09-09:
@bachya , I have never used this VerneMQ, so I don't know how to help with this.
On my Mosquitto, I have ACLs for the whole topic, like this:
```
user rtlamruser
topic readwrite homeassistant/sensor/rtlamr/#
topic readwrite rtlamr/#
```

With this configuration, the rtlamr user is allowed to send the auto-discover packages to HA and update their status.

Having one single user with RW rights to anything is the same as not having ACLs at all. If security is a concern (what should be), every client should have their own user with rights to specific topics.

You have to read the manual for your MQTT broker to have the right configuration in place.
This cannot be solved anywhere else.

allangood, 2021-09-09:
I am closing this as none of them are bug in the code.

Open a new issue or reopen this one if you have further comments.

Thank you for your feedback!

closed 2021-09-09 by allangood

ponchohoncho, 2021-09-09:
I'm still having an issue with meter readings. I got homeassistant autodiscover messages showing up in Mosquitto, but meter readings aren't coming in. Readings all showed up beautifully when DEBUG was on, but now the docker logs are full of this

```
listening...
RTLAMR started with PID 29962
Use the device argument 'rtl_tcp=127.0.0.1:1234' in OsmoSDR (gr-osmosdr) source
to receive samples in GRC and control rtl_tcp parameters (frequency, gain, ...).
bye!
RTL_TCP started with PID 29967
listening...
RTLAMR started with PID 29973
Use the device argument 'rtl_tcp=127.0.0.1:1234' in OsmoSDR (gr-osmosdr) source
to receive samples in GRC and control rtl_tcp parameters (frequency, gain, ...).
bye!
RTL_TCP started with PID 29977
listening...
RTLAMR started with PID 29983
Use the device argument 'rtl_tcp=127.0.0.1:1234' in OsmoSDR (gr-osmosdr) source
to receive samples in GRC and control rtl_tcp parameters (frequency, gain, ...).
bye!
RTL_TCP started with PID 29988
listening...
RTLAMR started with PID 29994
Use the device argument 'rtl_tcp=127.0.0.1:1234' in OsmoSDR (gr-osmosdr) source
to receive samples in GRC and control rtl_tcp parameters (frequency, gain, ...).
bye!
RTL_TCP started with PID 29999
listening...
RTLAMR started with PID 30005
Use the device argument 'rtl_tcp=127.0.0.1:1234' in OsmoSDR (gr-osmosdr) source
to receive samples in GRC and control rtl_tcp parameters (frequency, gain, ...).
bye!
RTL_TCP started with PID 30010

```

allangood, 2021-09-09:
@ponchohoncho how your configuration looks like and how are you running the container (i.e. the command you used to run the container)?

ponchohoncho, 2021-09-09:
thanks for your help @allangood ! here's my docker-compose entry

````
  rtlamr:
    container_name: rtlamr
    image: allangood/rtlamr2mqtt
    restart: unless-stopped
    devices:
      - /dev/bus/usb/001/003
    volumes:
      - ./rtlamr/rtlamr2mqtt.yaml:/etc/rtlamr2mqtt.yaml:ro
    #environment:
    #  - DEBUG=true
    #  - RTL_MSGTYPE="all"
````

and my rtlarm2mqtt.yaml

````
# (Optional section)
general:
  # Sleep for this amount of seconds after one successful of every meter
  # Set this to 0 (default) to disable it
  sleep_for: 0

# (Required section)
mqtt:
  host: 192.168.1.xx
  user: myusername
  password: mysecurepass
  ha_autodiscovery: true
  ha_autodiscovery_topic: homeassistant

# (Optional)
custom_parameters:
  rtltcp: "-s 2048000"
  # ***DO NOT ADD -msgtype, -filterid nor -protocol parameters here***
  rtlamr: "-unique=true -symbollength=7"

meters:
  - id: 65xxxx06
    name: meter_electric
    protocol: scm
    unit_of_measurement: kWh
    icon: mdi:gauge
  - id: 38xxxx38
    name: meter_gas
    protocol: scm
    unit_of_measurement: 'ft\uB003'
    icon: mdi:gauge
  - id: 19xxxx71
    name: meter_water
    protocol: scm
    unit_of_measurement: '\33A5'
    icon: mdi:gauge
````

hassio runs on a vm on another host, and rtlarm2mqtt runs on a docker host running ubuntu. i have a few other containers on this host that do mqtt and sdr (frigate, rtl_433) and they all work, so i know that communication, mqtt, etc all work between the two hosts/networks. i have one sdr device for rtl_433 and another one for rtlamr2mqtt.

allangood, 2021-09-09:
@ponchohoncho do you have ACLs on your MQTT broker?

ponchohoncho, 2021-09-09:
No, no ACLs, just the one user/pass

allangood, 2021-09-10:
@ponchohoncho , looks like the process is having problems to open the device.
Could you change the device section of your compose files to:
```
devices:
      - /dev/bus/usb
```

Also, does the `lsusb` command output your USB SDR device?

And last, just to be sure, all those IP, password, username and meters IDs that looks redacted are correct on your configuration?

ponchohoncho, 2021-09-10:
I think part of the problem is that it's a crapshoot to determine which container gets which device on reboot. I tried specifying the exact usb device ID in the devices part of docker-compose, but that wasn't perfect.

all the IPs, users, pass, etc are correct, just redacted for posting here.

both sdr sticks do show up in lsusb and rtl_test on the host. i have used the -d flag on rtl_433 to have it select a specific stick. i'll see if I can use custom_parameters to pass the -d flag to rtl_tcp in your container.

I think that this is an issue on my end with one container or the other not being able to use the hardware it wants. i'll keep messing with it.

it would be very helpful if there were more error messages caught from the underlying programs (paho-mqtt, rtl_tcp, rtlamr, etc) and passed up into your container's logs to debug things like this.

I think that your current DEBUG mode is more like a "discovery" mode, one to help someone narrow down what devices are available, their IDs, protocols, etc. the discovery mode wouldn't care about mqtt settings because it's not sending anything in. a discovery mode is super super helpful.

a real DEBUG mode would do everything the normal mode does (send to mqtt, autodiscover, fully parse all the yaml), just with more verbose info and errors, I think.

ponchohoncho, 2021-09-10:
even when i take down rtl433, and make sure that nothing is using either sdr stick, and rtlamr can take its pick of free devices to use, I still now get a log full of

````
RTLAMR started with PID 47
Use the device argument 'rtl_tcp=127.0.0.1:1234' in OsmoSDR (gr-osmosdr) source
to receive samples in GRC and control rtl_tcp parameters (frequency, gain, ...).
bye!
````

rtl_tcp or rtlamr is not liking something that's being passed to it

allangood, 2021-09-10:
@ponchohoncho , I agree the output could be more verbose. I am going to add a new `verbose` option to the configuration to address this.
Also, I need to handle better errors. I did the project initially just for myself and just assumed everything is working.

You can try this to debug the problem:
```
docker exec -ti rtlamr2mqtt /bin/ash
```
This will give you a shell inside the container. From there, you can execute:
```
rtl_tcp &
rtlamr -msgtype=scm+
```

allangood, 2021-09-10:
@ponchohoncho , every time your machine reboots, a big probability of your USB devices would be associated with different USB devices. A way to fix this is to use ***udev rules***.
With the right udev rule you can "pin point" one device with a specific /dev/ file

ponchohoncho, 2021-09-11:
When I open a shell into rtlamrmqtt and run rtl_test, I see both USB sticks there. I have rtl_433 using the nesdrv4 with the "-d1" flag. but when I try to run "rtl_test -d 0", to use the stick I've designated for rtlamr2mqtt, I get this:

````
Found 2 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: rtlsdrv3pro
  1:  Realtek, RTL2832UHIDIR, SN: nesdrv4

Using device 0: Generic RTL2832U OEM
usb_claim_interface error -6
Failed to open rtlsdr device #0.
````

isn't this because the container is already using it?

ps shows me this
````
/tmp # ps
PID   USER     TIME  COMMAND
    1 root      0:00 python3 /usr/bin/rtlamr2mqtt.py
  425 root      0:00 /bin/ash
 1084 root      0:00 /usr/bin/rtl_tcp -d 0 -s 2048000
 1090 root      0:00 ps
````

i can't kill the rtl_tcp process because it's constantly dying and restarting. killing python3 drops me out of that shell.

I added the -d 0 flag in rtltcp commands in rtlamr2mqtt's config yaml. since rtl_tcp would try 0 default anyways, i get the same results i've been getting either way.

Thanks so much for your help through all of this!

allangood, 2021-09-12:
@ponchohoncho , yes, if you kill any of the process, the python script will spawn a new one. One way to work around this is setting the `sleep_for` parameter with something > 0. This will give you that amount of seconds to kill those process.

Looks like all problems you are facing is because of these 2 antennas and the USB "mess".
In this case, having udev rules is highly recommended.

ponchohoncho, 2021-09-16:
ok, I'm back with my learnings. In short, with rtl_sdr in docker, udev rules don't matter. what you pass in via docker devices is not the same as what librtlsdr sees, and librtlsdr doesn't understand when it tries to use the first available sdr device listed, but can't due to docker device permissions.

librtlsdr chooses a device based on the device index, and can't be told to pick a device based on a serial number or udev name. lsusb and rtl_test see all sdr devices plugged into the host, but only have permission to talk to the ones passed in via the device flag in docker run/compose.

there's decently in-depth conversation about it here: https://forums.balena.io/t/mount-specific-usb-into-each-container/5651

imagine a docker setup like this. two containers that use rtl_sdr, and two sdr devices. let's imagine that in rtl_test, you see that the device on 001/002 is listed as device 0, and 001/003 is listed as device 1.

```
rtlamr:
    devices:
      - /dev/bus/usb/001/002 #device 1
rtl433:
    devices:
      - /dev/bus/usb/001/003 #device 0
```

if you run lsusb in both containers, you will see both sdr devices listed. but if you try to use rtl_test -d 1 in rtlamr, it'll give you an error because it doesn't have the permissions to use that device.

so it comes down to knowing which device is listed as which index, and then using the -d flag in each container to assign a device to that container, while passing /dev/bus/usb to each container.

where i think i got most tripped up is that in DEBUG/LISTEN mode, your container ignores the rtl_tcp flags in the yaml, so I thought I was specifying -d 1, when it was using -d 0 by default, and it couldn't use -d 0 because either it didn't have permissions as per the device flags, or because the other rtl container was already using it.

i have it all running now happily, and it's sending my meter readings to HA via MQTT, and rtl_433 is using the other device and all is right with the world. my only suggestion would be to allow the specifying of rtl_tcp flags that are used in regular, listen, and debug modes, so we can specify which device index to use for the container as a whole regardless of the mode.

thank you so much for making such a great container, listening to everyone's feedback, and being so quick with the fixes!