allangood/rtlamr2mqtt #17: Having issues with invalid symbollength

seaverd, 2021-09-14:
I am trying to get this docker running and am having a few issues.

1. When I add the extra parameter -e LISTEN_ONLY=yes my docker container doesn't start.  Prior to you recent revisions, I was able to get this to work in the old syntax.
2. See log below.  Getting an invalid symbollength.  Without being able to get the container to listen only mode I can't give you an example of the output my SDR is receiving.

I have also attached a copy of my .yaml file for config settings.  

[rtlamr2mqtt.txt](https://github.com/allangood/rtlamr2mqtt/files/7163240/rtlamr2mqtt.txt)

Thanks,
Dan

`RTLAMR started with PID 13
today at 11:19:33 AM 11:19:33.312989 flags.go:106: Environment variable "RTLAMR_VERSION" failed to override flag "version" with value "v0.9.1": "strconv.ParseBool: parsing \"v0.9.1\": invalid syntax"
today at 11:19:33 AM 11:19:33.313075 flags.go:124: invalid symbollength
today at 11:19:33 AM Sleep_for defined, time to sleep!
today at 11:19:33 AM Terminating all subprocess...
today at 11:19:33 AM Signal caught, exiting!
today at 11:19:33 AM Use the device argument 'rtl_tcp=127.0.0.1:1234' in OsmoSDR (gr-osmosdr) source
today at 11:19:33 AM to receive samples in GRC and control rtl_tcp parameters (frequency, gain, ...).
today at 11:19:33 AM bye!
today at 11:19:33 AM RTL_TCP terminated.
today at 11:19:33 AM RTLAMR terminated.
today at 11:19:33 AM Sleeping for 300 seconds, see you later...
today at 11:24:33 AM Sending message to MQTT:
today at 11:24:33 AM  > hostname => 192.168.10.50
today at 11:24:33 AM  > port => 11883
today at 11:24:33 AM  > username => seaverd
today at 11:24:33 AM  > password => *** REDACTED ***
today at 11:24:33 AM  > topic => rtlamr/status
today at 11:24:33 AM  > payload => online
today at 11:24:33 AM  > qos => 0
today at 11:24:33 AM  > retain => True
today at 11:24:33 AM RTL_TCP started with PID 17
today at 11:24:33 AM Found 1 device(s):
today at 11:24:33 AM   0:  Realtek, RTL2838UHIDIR, SN: 00000001
today at 11:24:33 AM
today at 11:24:33 AM Using device 0: Generic RTL2832U OEM
today at 11:24:33 AM Found Rafael Micro R820T tuner
today at 11:24:33 AM [R82XX] PLL not locked!
today at 11:24:34 AM Tuned to 100000000 Hz.
today at 11:24:34 AM listening...
today at 11:24:35 AM RTLAMR started with PID 22
today at 11:24:35 AM 11:24:35.489213 flags.go:106: Environment variable "RTLAMR_VERSION" failed to override flag "version" with value "v0.9.1": "strconv.ParseBool: parsing \"v0.9.1\": invalid syntax"
today at 11:24:35 AM 11:24:35.489294 flags.go:124: invalid symbollength
today at 11:24:35 AM Sleep_for defined, time to sleep!
today at 11:24:35 AM Terminating all subprocess...
today at 11:24:35 AM Signal caught, exiting!
today at 11:24:35 AM Use the device argument 'rtl_tcp=127.0.0.1:1234' in OsmoSDR (gr-osmosdr) source
today at 11:24:35 AM to receive samples in GRC and control rtl_tcp parameters (frequency, gain, ...).
today at 11:24:35 AM bye!
today at 11:24:35 AM RTL_TCP terminated.
today at 11:24:35 AM RTLAMR terminated.
today at 11:24:35 AM Sleeping for 300 seconds, see you later...
`

allangood, 2021-09-14:
Accordingly to the RTLAMR documentation, the lowest value for `-symbollength` is 8, on your configuration you have it set to 7, what it the cause of the problem.

https://github.com/bemasher/rtlamr/wiki/Configuration

A fix for the LISTEN mode is on the way!

Thank you for reporting!

seaverd, 2021-09-14:
OK, the 7 for symbol length came from the configuration sample on the readme.md page.  So I suggest that be revised to 8.

Thanks for your help!
Dan

allangood, 2021-09-14:
Ah, ok, I see the problem now.

I've made the changes in the README file and included a more detailed description of all parameters.

Thank you again.

Fixed with PR #18

closed 2021-09-15 by allangood