allangood/rtlamr2mqtt #25: Listen Only Logs spammed

mlebaugh, 2021-09-24:
Listen only logs are being spammed with messages like:

ll*, now xxx


ll+, now 360

ll+, now 361

ll+, now 362

ll+, now 363

ll+, now 364

ll+, now 365

ll+, now 366

ll+, now 367

I'm currently logging the output to syslog from this so I can try to identify my "wake only" gas/water meters. This is spamming the logs, making it difficult to sort through.

Additionally I'd like the option to allow it to listen AND log meters being set in the yaml to be sent to home assistant. My power meter is easily identified, but i have yet to see my gas/water. After looking into it I think they only send if the meter is being read, which could only be monthly. 

Thanks for the excellent container!

allangood, 2021-09-24:
Yes, this behaviour happens when no protocol is specified.
You can specify multiple protocols using comma-separated list of protocols.

More here:
`-msgtype section`
https://github.com/bemasher/rtlamr/wiki/Configuration

allangood, 2021-09-24:
@mlebaugh , the "LISTEN_ONLY" mode was created to make it easy for people to find their meters ID without having to configure anything. It is a quick way to sniff the network.

If you want more verbose messages, change the default verbosity level `verbosity: debug` in the configuration file.
What I can do to help with your user case is to add a `no_filter_id` parameter to listen for all IDs instead of only the listed IDs in the configuration.