allangood/rtlamr2mqtt #27: configuration option to specify the spectrum (tcp) server address

lymanbuttler, 2021-10-03:
When rtlamr runs within the container, it looks for the tcp server at 127.0.0.1:1234. Would it be possible to add a configuration option in the yaml to allow setting a custom address? I have a non-Dockerized instance of rtl_tcp but this is the perfect bridge to Home Assistant so I hope I can get them to talk to each other. Thanks!

jonbloom, 2021-10-06:
I've opened a PR to address this here: https://github.com/allangood/rtlamr2mqtt/pull/28

allangood, 2021-10-06:
Hi @jonbloom , first of all thank you for your contribution!

The configuration file has a section `custom_parameters` for this type of user case usage.
You can achieve what you want writing something like this:
```
custom_parameters:
  rtlamr: "-server=1.2.3.4:1234"
```
The main idea of this container is to have everything in one single place (rtl_tcp and rtlamr). This PR is to solve a very specific use case that can be achieved without any modification in the code.

allangood, 2021-10-10:
Code merged. Thank you @jonbloom

closed 2021-10-10 by allangood