allangood/rtlamr2mqtt #42: 'formatted_reading' or 'formated_reading'

ocean75, 2021-11-26:
Hello,

After the recent update 1.4.0 I am receiving the below error message when messages are sent from my water and gas meter. No information is forwarded to the MQTT broker. 
Reviewing rtlamr2mqtt.py it appears two different spellings ('formatted_reading' and 'formated_reading') have been used in the code. Any suggestion for how to correct this? 
Awesome product by the way. Thank you.

-------
{"Time":"2021-11-26T15:57:04.648837908Z","Offset":0,"Length":0,"Type":"R900","Message":{"ID":XXXXXXXXX,"Unkn1":131,"NoUse":32,"BackFlow":0,"Consumption":1135248,"Unkn3":0,"Leak":1,"LeakNow":0}}
Meter "XXXXXXXXX" - Consumption 113524.8. Sending value to MQTT.
Traceback (most recent call last):
  File "/usr/bin/rtlamr2mqtt.py", line 384, in <module>
    msg_payload = formatted_reading
NameError: name 'formatted_reading' is not defined. Did you mean: 'formated_reading'?
-------

ocean75, 2021-11-26:
After some trial and error I figured it out. Removed one of the 't' in 'formated_reading' on line 384 in rtlamr2mqtt.py, and things were up running again. :-)

Original rtlamr2mqtt.py
384 - msg_payload = formatted_reading
Modified rtlamr2mqtt.py
384 - msg_payload = formated_reading

allangood, 2021-11-27:
Thank you for your time and contribution!
This will be fixed with the next PR.

closed 2021-11-27 by allangood