allangood/rtlamr2mqtt #48: Meter readings are off by 10x since upgrade from 1.3.3

drmulligan, 2021-12-07:
I am using the rtlamr2mqtt add-on and missed that there have been a bunch of updates since 1.3.3.  I upgraded today and now my gas meter is off by a factor of 10.  My gas meter reports in 10's of ft³ which I set format: '#######0' to account for this.  After upgrading my gas meter readings are 1/10 of what they should be.

My config:  https://gist.github.com/drmulligan/b9c648e0be2048977e9480719c498cd0

Logs: https://gist.github.com/drmulligan/e372d3758112dc7e17bb767a83a2fe20

In the case logged above the consumption should be 8293150 ft³ but it's being reported as 829315 ft³.

drmulligan, 2021-12-07:
In case it is helpful, I have a test HA OS instance with an older build of rtlamr2mqtt addon with the same configuration which is still reporting the correct consumption value.  It is newer than my production instance but I can't tell you how much newer because of the recently fixed addon version number issue.   I can provide logs from it or dig into it to find what version it really is.

allangood, 2021-12-07:
Hi @drmulligan , thank you for you time reporting this issue!
I know exactly where the problem is and I am working to fix it ASAP.

drmulligan, 2021-12-07:
The value reported by sensor.gas_meter was 829315 at the time.  I will have to figure out how to subscribe to the right MQTT topic and get back to you regarding actual values being published.

allangood, 2021-12-07:
Bug confirmed! I just found what is causing this issue.

drmulligan, 2021-12-07:
I really don't know Python at all but I do know several other languages.  However I looked at the code briefly and I thought it looked like you rolled your own formatting code.  In Java and other languages there are libraries which provide formatting options like you created with rich feature sets. Plug a number, or string containing a number, in and a nicely formatted string comes out.  '#', '.', and '0' are all supported formatting characters in the libraries I am familiar with, along with many others.  

If this is all very obvious I am sorry, I don't mean to be offensive.  I really need to learn Python one of these days.

Where is the best place to discuss your addon?  I am a HA newb and I have questions and ideas but they are likely nonsense so I don't want to raise feature requests just yet.

allangood, 2021-12-07:
The problem was on the formatting code but on how `value_template` get the value.
This change came with a big PR with a lot of nice changes and I haven't noticed.

I am still putting this project on track and a proper way to test it is on my TODO list.

Feel free to open feature requests! I have a lot to discuss, just don't know how to reach everybody that is using it! :)
One thing I would like to talk about is how to expose the Linear Regression (anomaly detection) in a good way... 

Thank you for you time to report!

drmulligan, 2021-12-07:
My gas meter readings are back to normal.  Thank you.

closed 2021-12-07 by drmulligan