allangood/rtlamr2mqtt #58: rtlamr2mqtt not working after most recent updates

seaverd, 2021-12-10:
I am getting the errors below when attempting to run the most recent docker for rtlamr2mqtt.  Everything was working fine before the most recent updates. 

Any suggestions?

Thanks,
Dan

`Traceback (most recent call last):
File "/usr/lib/python3/dist-packages/numpy/core/__init__.py", line 22, in <module>
from . import multiarray
File "/usr/lib/python3/dist-packages/numpy/core/multiarray.py", line 12, in <module>
from . import overrides
File "/usr/lib/python3/dist-packages/numpy/core/overrides.py", line 7, in <module>
from numpy.core._multiarray_umath import (
ImportError: libblas.so.3: cannot open shared object file: No such file or directory

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
File "/usr/bin/rtlamr2mqtt.py", line 18, in <module>
import numpy as np
File "/usr/lib/python3/dist-packages/numpy/__init__.py", line 140, in <module>
from . import core
File "/usr/lib/python3/dist-packages/numpy/core/__init__.py", line 48, in <module>
raise ImportError(msg)
ImportError:

IMPORTANT: PLEASE READ THIS FOR ADVICE ON HOW TO SOLVE THIS ISSUE!

Importing the numpy C-extensions failed. This error can happen for

many reasons, often due to issues with your setup or how NumPy was
installed.

We have compiled some common reasons and troubleshooting tips at:

https://numpy.org/devdocs/user/troubleshooting-importerror.html

Please note and check the following:

* The Python version is: Python3.9 from "/usr/bin/python3"

* The NumPy version is: "1.19.5"

and make sure that they are the versions you expect.
Please carefully study the documentation linked above for further help.

Original error was: libblas.so.3: cannot open shared object file: No such file or directory`

seaverd, 2021-12-12:
> Could you paste your docker run command or the compose?
> If you are using the HA-Addon, could you try to rebuild it, please?

Here is the docker run command:

docker run -d --name='rtlamr2mqtt' --net='bridge' --privileged=true -e
TZ="America/New_York" -e HOST_OS="Unraid" -v
'/mnt/user/appdata/rtlamr2mqtt/':'/etc':'rw' -v
'/mnt/user/appdata/rtlamr2mqtt/history':'/var/lib/rtlamr2mqtt':'rw'
--device='/dev/bus/usb/003/019:/dev/bus/usb/003/019' --restart
unless-stopped 'allangood/rtlamr2mqtt'

 I run home assistant in docker, so there is no HA-Addon.  I am running on
a linux Unraid server...so they have a gui where you can create a template
for each docker.  Unraid then pulls the values from the template to then
create the docker run statement above.  I have been using rtlamr2mqtt for a
couple of months and it has worked perfectly up through tagged version
1.4.1.    When updating to your current version and before writing my
github issue I did not notice that you added a second volume mapping.  I
have added that to my docker run but still get the same error.  If I pull
version 1.4.1 by adding the tag at the end of the repository, the old
version runs and I have no issues.

Let me know if you need anything else.

Dan

allangood, 2021-12-12:
Hi @seaverd , the problem seems to be the ` -v '/mnt/user/appdata/rtlamr2mqtt/':'/etc'`.
The option this way, you are mounting the whole `/etc` into an empty directory on your host machine. Try to mount only and only the `rtlamr2mqtt.yaml` config file, like this:
`-v '/mnt/user/appdata/rtlamr2mqtt/rtlamr2mqtt.yaml':'/etcrtlamr2mqtt.yaml'`

seaverd, 2021-12-15:
Tried what you suggested, mounting just the file worked!  
Thanks - will close issue.

closed 2021-12-15 by seaverd