allangood/rtlamr2mqtt #61: NumPy error after recent Docker update

lymanbuttler, 2021-12-18:
I can't pinpoint exactly when this started, but updating Docker is the only thing I can think of that's changed.

```
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/numpy/core/__init__.py", line 22, in <module>
    from . import multiarray
  File "/usr/lib/python3/dist-packages/numpy/core/multiarray.py", line 12, in <module>
    from . import overrides
  File "/usr/lib/python3/dist-packages/numpy/core/overrides.py", line 7, in <module>
    from numpy.core._multiarray_umath import (
ImportError: libblas.so.3: cannot open shared object file: No such file or director
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/usr/bin/rtlamr2mqtt.py", line 18, in <module>
    import numpy as np
  File "/usr/lib/python3/dist-packages/numpy/__init__.py", line 140, in <module>
    from . import core
  File "/usr/lib/python3/dist-packages/numpy/core/__init__.py", line 48, in <module>
    raise ImportError(msg)
ImportError: 
IMPORTANT: PLEASE READ THIS FOR ADVICE ON HOW TO SOLVE THIS ISSUE!
Importing the numpy C-extensions failed. This error can happen for
many reasons, often due to issues with your setup or how NumPy was
installed.
We have compiled some common reasons and troubleshooting tips at:
    https://numpy.org/devdocs/user/troubleshooting-importerror.html
Please note and check the following:
  * The Python version is: Python3.9 from "/usr/bin/python3"
  * The NumPy version is: "1.19.5"
and make sure that they are the versions you expect.
Please carefully study the documentation linked above for further help.
Original error was: libblas.so.3: cannot open shared object file: No such file or directory
```

allangood, 2021-12-19:
This seems to be a duplicated of #58

lymanbuttler, 2021-12-19:
Oops, yes it is. Thanks!

closed 2021-12-19 by lymanbuttler