allangood/rtlamr2mqtt #62: Receiver context cancelled: i/o timeout

stu1811, 2021-12-20:
Using HA Supervised on Ubuntu. rtlamr2mqtt version 1.5.2. I tried "rebuiding" the addon without success. This used to work. Maybe a HA Core update broke it?

```
Trying to start RTL_TCP: /usr/bin/rtl_tcp -s 2048000
RTL_TCP started with PID 73365
RTL_TCP is ready to receive connections!
server: 127.0.0.1:1234, host: 127.0.0.1, port: 1234
Attempting to tune rtl_tcp to a different freq to shake things up
Successfully tickled rtl_tcp
Trying to start RTLAMR: /usr/bin/rtlamr -msgtype=r900,scm -format=json -filterid=1563533180,2691258 -unique=true
RTLAMR started with PID 73372
09:55:55.988557 decode.go:45: CenterFreq: 912600155
09:55:55.989476 decode.go:46: SampleRate: 2359296
09:55:55.989493 decode.go:47: DataRate: 32768
09:55:55.989496 decode.go:48: ChipLength: 72
09:55:55.989498 decode.go:49: PreambleSymbols: 32
09:55:55.989500 decode.go:50: PreambleLength: 4608
09:55:55.989502 decode.go:51: PacketSymbols: 116
09:55:55.989504 decode.go:52: PacketLength: 16704
09:55:55.989509 decode.go:59: Protocols: r900,scm
09:55:55.989512 decode.go:60: Preambles: 00000000000000001110010101100100,111110010101001100000
09:55:55.989515 main.go:124: GainCount: 29
09:56:00.990892 main.go:343: Receiver context cancelled.
09:56:00.991080 main.go:320: read tcp 127.0.0.1:38638->127.0.0.1:1234: i/o timeout
io.ReadFull
main.(*Receiver).Run.func2
	/go/pkg/mod/github.com/bemasher/rtlamr@v0.9.3/main.go:181
runtime.goexit
	/usr/local/go/src/runtime/asm_amd64.s:1581
Sleep_for defined, time to sleep!
Terminating all subprocess...
Kill process called.
Killing RTL_TCP...
Killed.
Killing RTLAMR...
Killed in the first attempt.
```

Configuration:
```
general:
  sleep_for: 60
  verbosity: debug
mqtt:
  host: core-mosquitto
  port: 1883
  user: ITS_A_ME_MARIO
  password: PASSWORD
  ha_autodiscovery: true
  ha_autodiscovery_topic: homeassistant
custom_parameters:
  rtltcp: '-s 2048000'
  rtlamr: '-unique=true'
meters:
  - id: 1563533180
    protocol: r900
    name: water_meter
    format: '#######.#'
    unit_of_measurement: ft³
    icon: mdi:water-pump
  - id: 2691258
    protocol: scm
    name: gas_meter
    format: '####.##'
    unit_of_measurement: ft³
    icon: mdi:gas-cylinder
```

stu1811, 2021-12-20:
I restarted the host and now it's working.... I'm going to leave this open for a few days then close.

allangood, 2021-12-20:
This sometimes happens on my RPi as well. When it does, I either reboot or unplug and plug the USB dongle back.

closed 2021-12-21 by allangood