allangood/rtlamr2mqtt #73: MQTT Sensor Unavailable

stu1811, 2022-01-17:
Recently my sensor in mqtt is unavailable. It may be related to issue #71 

I'm using 1.5.5 in HA. My sensor is showing up as unavailable. I see two topic published from the rtlamr2mqrt logs. I used the mqtt integration to manually subscribe to these topics and here is what I get back.
topic: `homeassistant/sensor/rtlamr/gas_meter/config`
```
{
    "name": "gas_meter",
    "unique_id": "2691258",
    "unit_of_measurement": "ft³",
    "icon": "mdi:gas-cylinder",
    "device_class": null,
    "availability_topic": "rtlamr/status",
    "state_class": "total_increasing",
    "state_topic": "rtlamr/2691258/state",
    "value_template": "{{ value_json.Message.Consumption | float }}",
    "json_attributes_topic": "rtlamr/2691258/state",
    "json_attributes_template": "{{ value_json.Message | tojson }}"
}
```

topic: `rtlamr/2691258/state` and I get the following results:
```
{
    "Time": "2022-01-16T20:48:29.30570082-05:00",
    "Offset": 0,
    "Length": 0,
    "Type": "SCM",
    "Message": {
        "ID": 2691258,
        "Type": 12,
        "TamperPhy": 3,
        "TamperEnc": 0,
        "Consumption": "3035.54",
        "ChecksumVal": 41625,
        "Predicted": 3035.5398942749307,
        "Anomaly": false
    }
}
```

My config file:
```
general:
  sleep_for: 60
  verbosity: debug
  tickle_rtl_tcp: false
  debug: true
mqtt:
  host: core-mosquitto
  port: 1883
  user: USERNAME
  password: PASSWORD
  ha_autodiscovery: true
  ha_autodiscovery_topic: homeassistant
custom_parameters:
  rtltcp: '-s 2048000'
  rtlamr: '-unique=true'
meters:
  - id: 2691258
    protocol: scm
    name: gas_meter
    format: '####.##'
    unit_of_measurement: ft³
    icon: mdi:gas-cylinder
```

Do I now need to add the mqtt topic to configuration.yaml? ha_autodiscovery is enabled so I didn't think it was necessary.

allangood, 2022-01-17:
Thank you for your detailed report!
This was a bug introduced by #66 . Home Assistant does not accept empty and now, doesn't accept null value for device_class.

The latest PR #74 should fix this.
Please, pull the latest version to fix this issue.

Sorry for the inconvenience.

stu1811, 2022-01-17:
New version is working. Thanks for the quick fix.

closed 2022-01-17 by stu1811