allangood/rtlamr2mqtt #83: wont start, tons of py errors

sargonas, 2022-02-04:
The following is just repeated constantly in my logs, and the container won't start properly.
(System ran fine for months, but I turned off the container and let it set for the last few months. Just updated it and tried to bring it back online and experienced this.)

```
Traceback (most recent call last):
  File "/usr/bin/rtlamr2mqtt.py", line 255, in <module>
    config = load_config(sys.argv)
  File "/usr/bin/rtlamr2mqtt.py", line 134, in load_config
    return load_yaml_config(config_path)
  File "/usr/bin/rtlamr2mqtt.py", line 144, in load_yaml_config
    return yaml.safe_load(config_file)
  File "/usr/lib/python3/dist-packages/yaml/__init__.py", line 162, in safe_load
    return load(stream, SafeLoader)
  File "/usr/lib/python3/dist-packages/yaml/__init__.py", line 114, in load
    return loader.get_single_data()
  File "/usr/lib/python3/dist-packages/yaml/constructor.py", line 49, in get_single_data
    node = self.get_single_node()
  File "/usr/lib/python3/dist-packages/yaml/composer.py", line 39, in get_single_node
    if not self.check_event(StreamEndEvent):
  File "/usr/lib/python3/dist-packages/yaml/parser.py", line 98, in check_event
    self.current_event = self.state()
  File "/usr/lib/python3/dist-packages/yaml/parser.py", line 171, in parse_document_start
    raise ParserError(None, None,
yaml.parser.ParserError: expected '<document start>', but found '<block mapping start>'
  in "/usr/bin/rtlamr2mqtt.py", line 29, column 1
```

allangood, 2022-02-04:
Looks like the config file is not there and it is pointing to a directory - default behavior when `-v` is used to point to a file that doesn't exists.

sargonas, 2022-02-04:
yeah i saw those threads and checked that to make sure I didn't do something silly back in the day, but I can confirm I have 
`- /etc/rtlamr2mqtt.yaml:/etc/rtlamr2mqtt.yaml:ro` in my volume mapping.

that said, if that error is indicative for an almotst-certain config error, then it's possible even though the volume mapping is right and the files there, some weird esoteric issue is causing the file to still not load anyways, so I'll dive into the config file angle more.

allangood, 2022-02-04:
Double check if the configuration file exists, having defined in your compose file does not create a file, but a directory if the file does not exists

allangood, 2022-02-04:
I.e. what this command outputs:
```
cat /etc/rtlamr2mqtt.yaml
```

sargonas, 2022-02-04:
It's definitely there. It was the first thing I looked at to ensure that I didn't have a malformed entry that was maybe causing this at the start of my debugging. Right now I'm poking a few things with sticks to see if perhaps while it's there, and formatted correctly, if it's failing to be loaded for some other unknown reason.

sargonas, 2022-02-04:
fixed. though I don't know why. I committed the cardinal sin of two troubleshooting changes before testing. ;)

first i changed the permissions on the existing config file to be far more permissive, in case something changed to make it no longer accessible by docker since before. Additionally, rather than recreating and restarting the container continuously, I nuked all traces of the container entry and just fully re-deployed my entire homeassistant compose stack. Now I seem to be up again!

closed 2022-02-04 by sargonas