allangood/rtlamr2mqtt #84: state class exceeding 255 chars

sargonas, 2022-02-09:
So over night my gas sensor stopped working and my HA logs began to fill with a repeated error:

```
2022-02-09 15:03:16 ERROR (MainThread) [homeassistant.util.logging] Exception in message_received when handling msg on 'rtlamr/98546445/state': '{"Time": "2022-02-09T23:03:16.552376855Z", "Offset": 0, "Length": 0, "Type": "SCM+", "Message": {"FrameSync": 5795, "ProtocolID": 30, "EndpointType": 156, "EndpointID": 98546445, "Consumption": "60090", "Tamper": 3592, "PacketCRC": 29813, "Predicted": 60110.93482659571, "Anomaly": false}}'
homeassistant.exceptions.InvalidStateError: Invalid state encountered for entity ID: sensor.gas. State max length is 255 characters.
```

I'm not sure what changed to suddenly upset the state class limit, if it was my update to 2022.2.x or something else, but now my gas meter is no longer reporting any info, as it's state class has ~280 some odd characters in it.

Am I doing something wrong that's causing too much information to be dumped into the state, or was there a change in 2022.2.X that made it more sensitive to the output from rtlamr2mqtt that's making the data in mqtt no longer valid to parse?

allangood, 2022-02-10:
Probably it is a misconfiguratuon in the configuration file. Could you
paste it?

sargonas, 2022-02-10:
```
general:
  sleep_for: 300
  verbosity: info
  tickle_rtl_tcp: false

mqtt:
  host: IPADDRESS
  user: USERNAME
  password: PASSWORD
  ha_autodiscovery: true
  ha_autodiscovery_topic: homeassistant

meters:
  - id: 98546445
    protocol: scm+
    name: gas_meter
    unit_of_measurement: m³
    icon: mdi:gauge
    device_class: gas
    state_class: total_increasing
  - id: 69055255
    protocol: scm+
    name: water_meter
    unit_of_measurement: Gal
    icon: mdi:gauge
    state_class: total_increasing
```

For what it's worth, I haven't changed the file between when it worked fine and now, though I did update HA from 2022.2.1 to 2022.4,

allangood, 2022-02-10:
Could you paste the container output as well, please?

sargonas, 2022-02-10:
Sure thing:
```
[2022-02-09 23:53:45] Meter "98546445" - Consumption 60090. Sending value to MQTT.
[2022-02-09 23:53:45] Predicted reading: 60117.752114154864 - Actual reading: 60090
[2022-02-09 23:53:45] Grow rate avg: 3.474860335195531
[2022-02-09 23:53:45] Distance from prediction: -27.752114154864103
[2022-02-09 23:53:45] Threshold for anomaly: 60121.22697449006
[2022-02-09 23:53:45] Sending message to MQTT:
[2022-02-09 23:53:45]  > topic => rtlamr/98546445/state
[2022-02-09 23:53:45]  > payload => {"Time": "2022-02-09T23:53:45.565985156Z", "Offset": 0, "Length": 0, "Type": "SCM+", "Message": {"FrameSync": 5795, "ProtocolID": 30, "EndpointType": 156, "EndpointID": 98546445, "Consumption": "60090", "Tamper": 3592, "PacketCRC": 29813, "Predicted": 60117.752114154864, "Anomaly": false}}
[2022-02-09 23:53:45]  > retain => True
```

allangood, 2022-02-10:
I just pulled the latest 2020.2.5 version and I can't reproduce the error.

Could you change the unit_of_measurement of your gas configuration to m3
just to test?

This toe of error usually happens when we open a quote and forget to close
it.

Also, could you pull your HA to the latest version just in case?

Thank you

sargonas, 2022-02-10:
HA is already on 2022.2.5 (I updated it earlier tonight sometime between opening this issue when i was on 2.4 and now).

I just swapped the m³ to m3 as requested, but seems to have no difference. Full HA output log is as follows:

```
2022-02-09 22:47:16 ERROR (MainThread) [homeassistant.util.logging] Exception in message_received when handling msg on 'rtlamr/98546445/state': '{"Time": "2022-02-10T06:47:16.345665418Z", "Offset": 0, "Length": 0, "Type": "SCM+", "Message": {"FrameSync": 5795, "ProtocolID": 30, "EndpointType": 156, "EndpointID": 98546445, "Consumption": "60094", "Tamper": 3592, "PacketCRC": 43189, "Predicted": 60178.2622424704, "Anomaly": false}}'
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/components/mqtt/debug_info.py", line 42, in wrapper
msg_callback(msg)
File "/usr/src/homeassistant/homeassistant/components/mqtt/sensor.py", line 283, in message_received
self.async_write_ha_state()
File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 536, in async_write_ha_state
self._async_write_ha_state()
File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 660, in _async_write_ha_state
self.hass.states.async_set(
File "/usr/src/homeassistant/homeassistant/core.py", line 1357, in async_set
state = State(
File "/usr/src/homeassistant/homeassistant/core.py", line 1045, in __init__
raise InvalidStateError(
homeassistant.exceptions.InvalidStateError: Invalid state encountered for entity ID: sensor.gas. State max length is 255 characters.
```

allangood, 2022-02-10:
Let me debug this in the morning :)

sargonas, 2022-02-10:
sounds good! I just turned the container off in the meantime... I'm not stressing over it right now so take you time and let me know if I can do anything to help.

allangood, 2022-02-10:
My state is very long too, it has all attributes like yours. Just counted and it has 300 characters and no errors on HA.

Would be possible for you to get one payload of the MQTT message?
Go to:
`HA -> Configuration -> Devices & Services -> Integration -> MQTT -> Configuration -> Listen to a topic`
And use this as the topic:
`rtlamr/98546445/state`

sargonas, 2022-02-10:
Bummer, I really hoped there would be some kind of smoking gun there to at least point in a good direction, lol./

```
Message 0 received on rtlamr/98546445/state at 3:27 PM:
{
    "Time": "2022-02-10T23:27:14.91797549Z",
    "Offset": 0,
    "Length": 0,
    "Type": "SCM+",
    "Message": {
        "FrameSync": 5795,
        "ProtocolID": 30,
        "EndpointType": 156,
        "EndpointID": 98546445,
        "Consumption": "60162",
        "Tamper": 3592,
        "PacketCRC": 21919,
        "Predicted": 60324.92725004349,
        "Anomaly": false
    }
}
```

allangood, 2022-02-11:
Maybe the configuration topic then, use this as topic:
`homeassistant/sensor/rtlamr/gas_meter/config`

sargonas, 2022-02-11:
Apologies, the work day has friend my brain and it's not immediately clear to me what you are asking. Did you want me to grab the payload for that topic from HA in the same way? If so here ya go!
```
Message 0 received on homeassistant/sensor/rtlamr/gas_meter/config at 4:48 PM:
{
    "name": "gas_meter",
    "unique_id": "98546445",
    "unit_of_measurement": "m3",
    "icon": "mdi:gauge",
    "availability_topic": "rtlamr/status",
    "state_class": "total_increasing",
    "state_topic": "rtlamr/98546445/state",
    "value_template": "{{ value_json.Message.Consumption | float }}",
    "json_attributes_topic": "rtlamr/98546445/state",
    "json_attributes_template": "{{ value_json.Message | tojson }}",
    "device_class": "gas"
}
```

allangood, 2022-02-11:
Yes, this what I've asked! Thank you.

Well, I really cannot see anything wrong with anything you have sent...

I'm going to make a change to split state_value and attributes into 2
different topics. This is something I was planning to do in the future, but
it will fix the problem you're experiencing.

I will update this issue in a few hours and will let you know how to test
it out.

Thank you and sorry for the inconvenience.

sargonas, 2022-02-11:
That did the trick!

thanks so much!

allangood, 2022-02-11:
Thank you for reporting.

closed 2022-02-11 by allangood