allangood/rtlamr2mqtt #85: rtlamr failing to get any data from rtl_tcp

JeffreyFalgout, 2022-02-10:
At some point today (~4 hours ago), rtlamr2mqtt stopped publishing updates for my electric meter. This is the pattern I'm seeing in the logs:
```
[2022-02-09 18:24:32] Trying to start RTL_TCP: /usr/bin/rtl_tcp -s 2048000
[2022-02-09 18:24:32] RTL_TCP started with PID 9
[2022-02-09 18:24:37] RTL_TCP is ready to receive connections!
[2022-02-09 18:24:37] Trying to start RTLAMR: /usr/bin/rtlamr -msgtype=scm+ -format=json -filterid=1552644360 -unique=true
[2022-02-09 18:24:37] RTLAMR started with PID 14
[2022-02-09 18:24:37] 18:24:37.077545 decode.go:45: CenterFreq: 912600155
[2022-02-09 18:24:37] 18:24:37.078061 decode.go:46: SampleRate: 2359296
[2022-02-09 18:24:37] 18:24:37.078082 decode.go:47: DataRate: 32768
[2022-02-09 18:24:37] 18:24:37.078093 decode.go:48: ChipLength: 72
[2022-02-09 18:24:37] 18:24:37.078104 decode.go:49: PreambleSymbols: 16
[2022-02-09 18:24:37] 18:24:37.078114 decode.go:50: PreambleLength: 2304
[2022-02-09 18:24:37] 18:24:37.078124 decode.go:51: PacketSymbols: 128
[2022-02-09 18:24:37] 18:24:37.078134 decode.go:52: PacketLength: 18432
[2022-02-09 18:24:37] 18:24:37.078151 decode.go:59: Protocols: scm+
[2022-02-09 18:24:37] 18:24:37.078161 decode.go:60: Preambles: 0001011010100011
[2022-02-09 18:24:37] 18:24:37.078172 main.go:124: GainCount: 29
[2022-02-09 18:24:42] 18:24:42.083166 main.go:343: Receiver context cancelled.
[2022-02-09 18:24:42] 18:24:42.085118 main.go:320: read tcp 127.0.0.1:40362->127.0.0.1:1234: i/o timeout
[2022-02-09 18:24:42] io.ReadFull
[2022-02-09 18:24:42] main.(*Receiver).Run.func2
[2022-02-09 18:24:42] 	/go/pkg/mod/github.com/bemasher/rtlamr@v0.9.3/main.go:181
[2022-02-09 18:24:42] runtime.goexit
[2022-02-09 18:24:42] 	/usr/local/go/src/runtime/asm_arm64.s:1133
```

There's a 5 second deadline being added here: https://github.com/bemasher/rtlamr/blob/master/main.go#L174
And then the error's coming from just a couple lines further down: https://github.com/bemasher/rtlamr/blob/master/main.go#L181

JeffreyFalgout, 2022-02-10:
I tried setting `tickle_rtl_tcp: true` in case that's what that flag was designed to fix, and it didn't seem to help

JeffreyFalgout, 2022-02-10:
This might be an issue with my SDR. I reseated the USB dongle and that seems to have fixed the problem. I'm using one of [these](https://smile.amazon.com/gp/product/B008S7AVTC/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1) if anyone's curious :shrug:

closed 2022-02-10 by JeffreyFalgout

allangood, 2022-02-10:
I have exactly the same USB model and mine stops to work after a few days
too.

Sometimes a simple container restart fixes it, sometimes I need a reboot
and sometimes I have to remove the USB stick, wait it cool down and plug it
back in. Looks Look like a hardware issue. Not sure if it is on my OrangePi
or the USB stick.

kspearrin, 2022-04-01:
@allangood I have the same receiver and am having the same issue. At first it ran fine for several weeks, but now after a few ours this same issue starts occurring again. If I remove the USB dongle and plug it back in it starts working again for a few hours.

I guess this is a hardware issue and I should look for a better device?

allangood, 2022-04-01:
@kspearrin , yes, this is the same behavior I have on my side. I manage to get it working for a few months non-stop after I added a cooler/fan to my  Pi (Orange Pi, actually).

I am really not sure what is the cause of this issue, but keeping the computer and the USB dongle cool helped.

I am using one like [this](https://www.amazon.ca/GeeekPi-Raspberry-Radiator-Aluminum-Heatsinks/dp/B07C9C99RM/ref=sr_1_27?crid=2F24R668AFMAV&keywords=pi+cooler&qid=1648827259&sprefix=pi+cooler%2Caps%2C121&sr=8-27)

JeffreyFalgout, 2022-04-01:
FWIW, my "solution" has been to add an automation that alerts me when it breaks. My Pi has a [PoE hat](https://www.raspberrypi.com/products/poe-hat/) on it that has a built in fan. Maybe adding more cooling would help the dongle be more stable, but I don't get the alert _too_ often :shrug: 

```
alias: electric_meter.alert
description: ''
trigger:
  - platform: state
    entity_id: sensor.electric_meter
    for:
      hours: 0
      minutes: 30
      seconds: 0
condition: []
action:
  - service: notify.jeff
    data:
      message: The electric meter hasn't updated in 30 minutes
mode: single
```

allangood, 2022-04-02:
I just sent a PR that may help with this situation (if it is not being caused by a hardware failure).
PR #101 

It has a bit more discussion and how to test before the next release on issue #98

kspearrin, 2022-05-16:
> @kspearrin , yes, this is the same behavior I have on my side. I manage to get it working for a few months non-stop after I added a cooler/fan to my Pi (Orange Pi, actually).
>
> I am really not sure [...]

@allangood Wanted to update you. I contacted the manufacturer complaining about the issue and they suggested that it is a power problem from the device I have it plugged into (in my case, a proxmox server). They suggested a powered USB hub. I bought [this one](https://www.amazon.com/gp/product/B00TPMEOYM/) on amazon and have been using it for over a week now without the issue occurring. Might be worth a try for you as well.