allangood/rtlamr2mqtt #100: Allow ability to configure TLS

bachya, 2022-03-31:
I notice from here:

https://github.com/allangood/rtlamr2mqtt/blob/1a2c69ffb0d068673a006c1b2bd14b005dca2a4e/rtlamr2mqtt.py#L73-L76

...that TLS is always assumed to be excluded. I host my MQTT broker behind TLS and would love to be able to utilize that here.

allangood, 2022-03-31:
Thank you for your feedback. This really should be addressed! :)
I will be working on it and release it in the next version.

allangood, 2022-04-06:
Added on version 1.7.0 by PR #104

JeffreyFalgout, 2022-04-06:
FWIW I'm not sure how necessary this is if you're using the automatic MQTT configuration. Presumably you get a local docker IP address to talk to, so the network traffic won't ever be leaving your computer

bachya, 2022-04-06:
@JeffreyFalgout That only applies to a very specific scenario. In my case, my rtlamr2mqtt and my MQTT broker are on different hosts.

bachya, 2022-04-07:
I can confirm that 1.7.0 implements this and that everything works. Thank you!

closed 2022-04-07 by bachya