allangood/rtlamr2mqtt #108: Unknown exception connecting to MQTT broker: [Errno 2] No such file or directory

dinki, 2022-04-08:
First, thanks for your work on this project.  I've been running rtlamr on my Linux server for a while and have finally gotten to a point where I wanted to switch it over to Docker.  This project looks like it fits my needs perfectly!

As noted above, I am getting an error when trying to use the docker container.  Here's the logs:

```
[2022-04-08 12:35:24] Sending message to MQTT:,
[2022-04-08 12:35:24]  > topic => rtlamr/status,
[2022-04-08 12:35:24]  > payload => offline,
[2022-04-08 12:35:24]  > retain => True,
[2022-04-08 12:35:24] Unknown exception connecting to MQTT broker: [Errno 2] No such file or directory,
[2022-04-08 12:35:25] Configured MQTT sender:,
[2022-04-08 12:35:25]  > hostname => 192.168.0.25,
[2022-04-08 12:35:25]  > port => 1883,
[2022-04-08 12:35:25]  > username => hass,
[2022-04-08 12:35:25]  > client_id => rtlamr2mqtt,
[2022-04-08 12:35:25]  > tls => {'ca_certs': '/etc/ssl/certs/ca-certificates.crt', 'certfile': '/etc/ssl/my_self_signed_cert.crt', 'insecure': True, 'keyfile': '/etc/ssl/my_self_signed_cert_key.key', 'tls_version': None, 'ciphers': None},
[2022-04-08 12:35:25] Sending message to MQTT:,
[2022-04-08 12:35:25]  > topic => rtlamr/status,
[2022-04-08 12:35:25]  > payload => online,
[2022-04-08 12:35:25]  > retain => True,
[2022-04-08 12:35:25] Unknown exception connecting to MQTT broker: [Errno 2] No such file or directory,
[2022-04-08 12:35:25] Trying to start RTL_TCP: /usr/bin/rtl_tcp -s 14448385,
[2022-04-08 12:35:25] RTL_TCP started with PID 7,
[2022-04-08 12:35:30] RTL_TCP is ready to receive connections!,
[2022-04-08 12:35:30] Trying to start RTLAMR: /usr/bin/rtlamr -msgtype=scm+ -format=json -filterid=14448385 -unique=true -symbollength=32,
[2022-04-08 12:35:30] RTLAMR started with PID 13,
[2022-04-08 12:35:30] 12:35:30.626183 decode.go:45: CenterFreq: 912600155,
[2022-04-08 12:35:30] 12:35:30.626264 decode.go:46: SampleRate: 1048576,
[2022-04-08 12:35:30] 12:35:30.626271 decode.go:47: DataRate: 32768,
[2022-04-08 12:35:30] 12:35:30.626276 decode.go:48: ChipLength: 32,
[2022-04-08 12:35:30] 12:35:30.626545 decode.go:49: PreambleSymbols: 16,
[2022-04-08 12:35:30] 12:35:30.626553 decode.go:50: PreambleLength: 1024,
[2022-04-08 12:35:30] 12:35:30.626557 decode.go:51: PacketSymbols: 128,
[2022-04-08 12:35:30] 12:35:30.626562 decode.go:52: PacketLength: 8192,
[2022-04-08 12:35:30] 12:35:30.626568 decode.go:59: Protocols: scm+,
[2022-04-08 12:35:30] 12:35:30.626572 decode.go:60: Preambles: 0001011010100011,
[2022-04-08 12:35:30] 12:35:30.626577 main.go:124: GainCount: 29,
[2022-04-08 12:46:38] Shutdown detected, killing process.,
[2022-04-08 12:46:38] Killing RTL_TCP...,
[2022-04-08 12:46:43] Killed.,
[2022-04-08 12:46:43] Killing RTLAMR...,
[2022-04-08 12:46:43] Killed in the first attempt.,
[2022-04-08 12:46:43] Graceful shutdown.,
```

and here's my rtlamr2mqtt.yaml file:

```
general:
  sleep_for: 0
  verbosity: debug
  tickle_rtl_tcp: false
  usb_reset: '002:004'

mqtt:
  host: 192.168.0.25
  # Path to CA certificate to use. Mandatory
  tls_ca: '/etc/ssl/certs/ca-certificates.crt'
  # Path to certificate file to use. Optional
  tls_cert: '/etc/ssl/my_self_signed_cert.crt'
  # TLS insecure: defaults to False
  tls_insecure: true
  # Certificate key file to use. Optional
  tls_keyfile: '/etc/ssl/my_self_signed_cert_key.key'
  user: 'hass'
  password: 'XXXXXXXXXXX'
  ha_autodiscovery: true
  ha_autodiscovery_topic: homeassistant

custom_parameters:
  rtltcp: "-s 14448385"
  rtlamr: "-unique=true -symbollength=32"

meters:
  - id: 14448385
    protocol: scm+
    name: test_meter
    format: "######.###"
    unit_of_measurement: "\u33A5"
    icon: mdi:gauge
    device_class: energy

```

and finally, my docker-compose entries:

```
  rtlamr:
    container_name: rtlamr2mqtt
    image: allangood/rtlamr2mqtt
    restart: unless-stopped
    depends_on:
      - rtl433
    devices:
      - /dev/bus/usb
    volumes:
      - /sharedfolders/appdata/rtlamr2mqtt/rtlamr2mqtt.yaml:/etc/rtlamr2mqtt.yaml:ro
      - /sharedfolders/appdata/rtlamr2mqtt/data:/var/lib/rtlamr2mqtt
 ```

It appears that the container is reading the config correctly but I'm not sure why it's having a hard time connecting to the MQTT broker.  I'm using it successfully with a handful of other containers.  This is, however, the first time I've tried using a user for connecting as my broker accepts anonymous connections.

Any help in troubleshooting this is most appreciated!

allangood, 2022-04-08:
Looks like the problem is the tls configuration. Are you using a MQTT
broker with TLS encryption? If not, just remove the tls configuration. If
yes, then you must point the configuration to existing files.

allangood, 2022-04-08:
I have to make some changes in the README to clarify when the tls options should be present in the config file.

jcollas, 2022-04-08:
I have the same issue and when I remove the tls configuration, the tls default configuration pops back up on restart.

dinki, 2022-04-08:
Thanks @allangood !  That did fix it.   I think what got me is the example file states:

```
  # Path to CA certificate to use. Mandatory
```

So I thought that it was necessary.  The container is up and appears to be active.  I do get the online message in HA but I am not getting meter readings.  I did run the command one the command line and know my meter number.  From that command I see:

```
[2022-04-08 12:03:10] {"Time":"2022-04-08T12:03:10.643465156Z","Offset":0,"Length":0,"Type":"SCM","Message":{"ID":14448385,"Type":4,"TamperPhy":0,"TamperEnc":2,"Consumption":1126,"ChecksumVal":48424}}
```

This is the config I'm using.  Am I doing something incorrectly?

```
custom_parameters:
  rtltcp: "-s 14448385"
  rtlamr: "-unique=true -symbollength=32"

meters:
  - id: 14448385
    protocol: scm+
    name: test_meter
    format: "######.###"
    unit_of_measurement: "\u33A5"
    icon: mdi:gauge
    device_class: energy

```

Apologies for the constant questioning.  I'm sure I'm overlooking something obvious but could certainly use a push in the right direction if possible.

allangood, 2022-04-08:
Yes, that bit it is very confuse in the README file... it is mandatory **only** if the tls is being used.
If you are not using TLS it is safe to remove the whole "tls_something"... I will push some changes to make this better.

Looking the output, your meter is using **SCM** and your configuration seems to be using **SCM+**

dinki, 2022-04-08:
Yep.  Started working as expected @allangood  .. I guess I'll push my luck one last time and ask how I can update the name and unit of measure.  I've made the change in the config file but HA is still seeing it with the test_meter name and the default units.

allangood, 2022-04-08:
Just restart the container:

`docker restart <container_name>`

dinki, 2022-04-08:
@allangood   Thank you.  That is what I've tried and it does not work.  Is it something with the persistence in MQTT?

allangood, 2022-04-08:
No, it is not related. Check the sensors again, you may have both there. The old and the new one. :)

dinki, 2022-04-08:
So I cheated.  I went and turned off autodiscovery and created the device in HA sensors.  This appears to be fine.  I blindly copied the example and I'm thinking about the format config.  My meter is old.  It has five little numbered wheels with an arrow that points to each.  The reading, as you can see above, gives that value.  I'm thinking that I need to change the format to match or just get rid of the entry altogether and that will give the whole integer value rather than the float that I have now, right?

dinki, 2022-04-08:
The actual daily consumption is the reading at midnight minus the current reading.  I'm guessing that's what the utility meter definitions (hourly/daily/monthly) calculate?

allangood, 2022-04-08:
This is not related to the issue anymore. Could open a discussion to ask for help instead?

Thank you.

closed 2022-04-08 by allangood

dinki, 2022-04-08:
Thanks for helping get this working.  Sorry went off topic at the end.  All the best to you!