allangood/rtlamr2mqtt #113: Breaking change from new update

markwoodward86, 2022-04-12:
Helloe there, i updated today and immediately ran into trouble. Im unable to remove the USB part of the config.

[2022-04-12 18:56:37] Configured MQTT sender:
[2022-04-12 18:56:37]  > hostname => core-mosquitto
[2022-04-12 18:56:37]  > port => 1883
[2022-04-12 18:56:37]  > username => homeassistant
[2022-04-12 18:56:37]  > client_id => rtlamr2mqtt
[2022-04-12 18:56:37]  > tls => None
[2022-04-12 18:56:37] Reseting USB device: /dev/bus/usb/005/002
Traceback (most recent call last):
  File "/usr/bin/rtlamr2mqtt.py", line 403, in <module>
    reset_usb_device(usb_reset)
  File "/usr/bin/rtlamr2mqtt.py", line 34, in reset_usb_device
    fd = open(filename, "wb")
FileNotFoundError: [Errno 2] No such file or directory: '/dev/bus/usb/005/002'

bcotton, 2022-04-13:
Can confirm. When I remove the `usb_reset` or the `rtltcp` options and hit save, they re-appear the config

Version | core-2022.2.9

allangood, 2022-04-13:
Pushed version 1.7.2 to fix this issue.
Set `usb_reset` parameter to a non existent or blank value to disable it.

But I recommend to have it enabled and pointing to your USB device.

allangood, 2022-04-13:
Could you try to pull the latest version (1.7.2) again to check if the problem persists?

Thank you

markwoodward86, 2022-04-13:
Hi Allan, thanks for following up. I tried a null or none value and got a different error (it was still attempting a reset). I ended up just finding the USB hardware and adding it to the config as you had originally intended. Im all good on my particular sticking point since this.

allangood, 2022-04-13:
Thank you @markwoodward86 for your feedback.

The blank/null/non-existent value fix was pushed a couple of minutes ago. Did you pushed the latest version before testing it?

markwoodward86, 2022-04-13:
Ah, i see. No i did not grab the latest one. I tried approximately 12 hours ago.

allangood, 2022-04-14:
Anyone could confirm if the problem is gone, please? :)

markwoodward86, 2022-04-15:
Hi Allan.

It turns out i do have the 1.7.2 version. 

I tried to remove the line 'usb_reset' entirely. This did not work - the config reverted back to include the line.
I tried None - this threw up the attached error.
I tired 'None' - this threw up the attached error.
I tried 'null' - this threw up the attached error.
I tried '009:009' (a false hardware location) - this threw up the attached error also but substitue the last comment with "no such file directory"
[image]
.

Im happy to try other arguments or variables if you like?

markwoodward86, 2022-04-15:
to add i tried false and 'false' too. Wouldn't let me save the config.

allangood, 2022-04-15:
I just pushed a fix to this release. Could you do a docker pull and test again, please?
Sorry about this release... :(

markwoodward86, 2022-04-15:
please dont apologise, i basically run home assistant solely for this add on. I appreciate your time and expertise. I, on the other hand am still learning and I dont think i can do a docker pull. Im running hassio on virtualbox. Im just a novice (but i know how to break things!)

allangood, 2022-04-15:
I believe if you uninstall and install the addon again it will pull the latest release.

allangood, 2022-04-18:
Thank you for all your help and feedback.

closed 2022-04-18 by allangood