allangood/rtlamr2mqtt #121: Add-on support for arm64

toddq, 2022-04-29:
HA says the add-on is not compatible with my processor.  I see that the Docker image build includes arm64, should the add-on config's `arch` be updated to include arm64 as well?  Or is there a reason that it wouldn't actually be suported?

allangood, 2022-04-29:
Hi @toddq , you are absolutely right, I should add this architecture to the supported list.
Thank you!

allangood, 2022-04-29:
For reference, these are the supported (by github actions) platforms now:
`platforms: linux/amd64,linux/i386,linux/arm64,linux/arm/v7,linux/arm/v6`

makeitworktech, 2022-04-29:
I think I'm having the same problem - is this fixable?

`22-04-29 16:59:33 WARNING (SyncWorker_4) [supervisor.store.data] Can't read /data/addons/git/6713e36e/rtlamr2mqtt-addon/config.json: value must be one of ['aarch64', 'amd64', 'armhf', 'armv7', 'i386'] @ data['arch'][3]. Got 'armv6'
value must be one of ['aarch64', 'amd64', 'armhf', 'armv7', 'i386'] @ data['arch'][4]. Got 'arm64'`

karepiu, 2022-04-30:
Exactly same here. Trying add-on rpi3 and same error as above

allangood, 2022-04-30:
I have added the wrong arch name: `arm64` instead of `aarch64` what caused this error.
Last commit fixed it.

Thank you for reporting!

toddq, 2022-05-02:
Thanks!  One remaining thing is to update the badge in the add-on Readme.  It's still reporting no aarch64, which will be confusing.

allangood, 2022-05-06:
Thank you for all your feedback and help!

closed 2022-05-06 by allangood