allangood/rtlamr2mqtt #123: usb device resets w/ remote rtl_tcp?

hume-github, 2022-05-04:
The code looks like it does usb resets regardless of whether or not rtl_tcp is local or remote.  Is that intentional?

allangood, 2022-05-05:
Absolute not intentional! :)
Thank you for pointing it out!

randysaz928, 2022-05-14:
The code appears to not check for whether remote rtl_tcp is used until pretty late, looks like in the meters filter setup section.  By then it has already found or guessed a rtlsdr and tried to reset it.  If you dont have a local radio it will also exit before getting to the remote rtl_tcp setup.
Commenting line 309 "    sys.exit(1)" should allow it to get to the setup of the meters, and the remote rtl_tcp setup when no local radio is present.  Albeit with errors about USB/devices being invalid.

My issue is that I have a local radio on HA that I use rtl433 and a remote rtl_tcp instance for rtlamr.  Weather station is in the back yard on 433 near my office and the HA server, and my water meter on 915 is the other direction toward the front of the house..

I suppose it is time for me to learn python... :(

allangood, 2022-05-14:
Yes, this problem was introduced with the USB reset functionality in version 2.0.0.

You can either use the previous version or wait a few days until I finish the fix.

I am organizing the code a bit and fixing this bug alone would make me redo a lot of things, that is why is taking more than usual to fix it.

allangood, 2022-05-17:
Small correction on #132 
Thank you @shbatm

closed 2022-05-18 by allangood