allangood/rtlamr2mqtt #134: MQTT config not accepting non local server

BigWebstas, 2022-05-19:
When I specify the host,port,user,password it still trys to grab the local HASS MQTT Addon which in my case doesnt exist causing it to bomb out
`[2022-05-19 11:34:49] RTLAMR2MQTT Starting...
[2022-05-19 11:34:49] Fetching default MQTT configuration from http://supervisor/services/mqtt
Traceback (most recent call last):
  File "/usr/bin/rtlamr2mqtt.py", line 286, in load_config
    resp.raise_for_status()
  File "/usr/local/lib/python3.9/site-packages/requests/models.py", line 960, in raise_for_status
    raise HTTPError(http_error_msg, response=self)
requests.exceptions.HTTPError: 400 Client Error: Bad Request for url: http://supervisor/services/mqtt
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/usr/bin/rtlamr2mqtt.py", line 372, in <module>
    config = load_config(sys.argv)
  File "/usr/bin/rtlamr2mqtt.py", line 297, in load_config
    except e:
NameError: name 'e' is not defined`

`general:
  sleep_for: 300
  verbosity: debug
  tickle_rtl_tcp: false
  device_id: single
mqtt:
  ha_autodiscovery: true
  ha_autodiscovery_topic: homeassistant
  base_topic: rtlamr
  tls_enabled: false
  tls_ca: /etc/ssl/certs/ca-certificates.crt
  tls_cert: /etc/ssl/my_self_signed_cert.crt
  tls_keyfile: /etc/ssl/my_self_signed_cert_key.key
  tls_insecure: true
  host: 10.2.0.140
  port: 1883
  user: rtl
  password: xxxxxx
custom_parameters:
  rtltcp: '-s 2048000'
  rtlamr: '-unique=true'
meters:
  - id: 123456789
    protocol: scm+
    name: gas_meter
    format: '####.###'
    unit_of_measurement: m³
    icon: mdi:gas_canister
    device_class: gas
`

allangood, 2022-05-19:
Hi @BigWebstas , this is a new use case for me :)
I am going to make a change to allow the supervised configuration to be overwritten by the YAML file.

Thank you for reporting.

randysaz928, 2022-05-26:
I have the same issue. Running current release of HA and rtlamr2mqtt v 2.1.1 and Mosquito running on a separate instance of HA with password protection and no ssl.   
If it finds a local instance of mqtt , i.e. core-mosquitto, it ignores the config for mqtt: host: user: password:   if it doesn't find a local instance it throws the "load_config except e: NameError: name 'e' is not defined" error...
On the brighter side, it is already Wednesday.  Happy Wednesday! :)

allangood, 2022-06-09:
Functionality added on #137

closed 2022-06-09 by allangood