allangood/rtlamr2mqtt #140: Unknown exception connecting to MQTT broker: tls_set() got an unexpected keyword argument 'tls_insecure'

bachya, 2022-06-09:
It looks like this:

https://github.com/allangood/rtlamr2mqtt/blob/f155beb975234b717a44519d1f30581038c2c326/rtlamr2mqtt.py#L136

...causes an the exception in the title. I'm guessing this refers to the `tls_set` method in `paho-mqtt`, whose [definition](https://github.com/eclipse/paho.mqtt.python/blob/9782ab81fe7ee3a05e74c7f3e1d03d5611ea4be4/src/paho/mqtt/client.py#L735) doesn't include that kwarg:

```python
def tls_set(self, ca_certs=None, certfile=None, keyfile=None, cert_reqs=None, tls_version=None, ciphers=None, keyfile_password=None):
```

allangood, 2022-06-09:
Wow, awesome explanation! :)

I will fix this in the next minor release!

Thank you very much!

allangood, 2022-06-13:
Change pushed to the dev tag.

You can test it using `allangood/rtlamr2mqtt:dev` as image.

Thank you for your help!

bachya, 2022-06-13:
I just spun up the `dev` Docker image and am still seeing the same error in the logs:

```
rtlamr2mqtt_1  | [2022-06-13 18:44:09] RTLAMR2MQTT Starting...
rtlamr2mqtt_1  | [2022-06-13 18:44:09] Using "/etc/rtlamr2mqtt.yaml" config file
rtlamr2mqtt_1  | [2022-06-13 18:44:09] RTL SDR Device 0bda:2838 found on USB port 002:002 - Index: 0
rtlamr2mqtt_1  | [2022-06-13 18:44:09] No USB device specified in the config file, using the first found.
rtlamr2mqtt_1  | [2022-06-13 18:44:09] Configured MQTT sender:
rtlamr2mqtt_1  | [2022-06-13 18:44:09]  > hostname => mqtt.mydomain.com
rtlamr2mqtt_1  | [2022-06-13 18:44:09]  > port => 8883
rtlamr2mqtt_1  | [2022-06-13 18:44:09]  > username => rtlamr
rtlamr2mqtt_1  | [2022-06-13 18:44:09]  > client_id => rtlamr2mqtt
rtlamr2mqtt_1  | [2022-06-13 18:44:09]  > base_topic => rtlamr
rtlamr2mqtt_1  | [2022-06-13 18:44:09]  > availability_topic => rtlamr/status
rtlamr2mqtt_1  | [2022-06-13 18:44:09]  > tls => {'ca_certs': '/etc/ssl/certs/ca-certificates.crt', 'certfile': None, 'keyfile': None, 'tls_insecure': True}
rtlamr2mqtt_1  | [2022-06-13 18:44:09] Reseting USB device: /dev/bus/usb/002/002
rtlamr2mqtt_1  | [2022-06-13 18:44:10] Reset sucessful.
rtlamr2mqtt_1  | [2022-06-13 18:44:10] Sending message to MQTT:
rtlamr2mqtt_1  | [2022-06-13 18:44:10]  > topic => rtlamr/status
rtlamr2mqtt_1  | [2022-06-13 18:44:10]  > payload => online
rtlamr2mqtt_1  | [2022-06-13 18:44:10]  > retain => True
rtlamr2mqtt_1  | [2022-06-13 18:44:10] Unknown exception connecting to MQTT broker: Client.tls_set() got an unexpected keyword argument 'tls_insecure'
rtlamr2mqtt_1  | [2022-06-13 18:44:10] Trying to start RTL_TCP: /usr/bin/rtl_tcp  -s 2048000
rtlamr2mqtt_1  | [2022-06-13 18:44:10] RTL_TCP started with PID 8
rtlamr2mqtt_1  | [2022-06-13 18:44:15] RTL_TCP is ready to receive connections!
rtlamr2mqtt_1  | [2022-06-13 18:44:15] Trying to start RTLAMR: /usr/bin/rtlamr -msgtype=scm,scm+ -format=json -filterid=51643176,101905796 -unique=true
rtlamr2mqtt_1  | [2022-06-13 18:44:15] RTLAMR started with PID 13
rtlamr2mqtt_1  | [2022-06-13 18:44:15] 18:44:15.257331 decode.go:45: CenterFreq: 912600155
rtlamr2mqtt_1  | [2022-06-13 18:44:15] 18:44:15.257703 decode.go:46: SampleRate: 2359296
rtlamr2mqtt_1  | [2022-06-13 18:44:15] 18:44:15.257810 decode.go:47: DataRate: 32768
rtlamr2mqtt_1  | [2022-06-13 18:44:15] 18:44:15.258063 decode.go:48: ChipLength: 72
rtlamr2mqtt_1  | [2022-06-13 18:44:15] 18:44:15.258257 decode.go:49: PreambleSymbols: 21
rtlamr2mqtt_1  | [2022-06-13 18:44:15] 18:44:15.258478 decode.go:50: PreambleLength: 3024
rtlamr2mqtt_1  | [2022-06-13 18:44:15] 18:44:15.258671 decode.go:51: PacketSymbols: 128
rtlamr2mqtt_1  | [2022-06-13 18:44:15] 18:44:15.258875 decode.go:52: PacketLength: 18432
rtlamr2mqtt_1  | [2022-06-13 18:44:15] 18:44:15.259069 decode.go:59: Protocols: scm,scm+
rtlamr2mqtt_1  | [2022-06-13 18:44:15] 18:44:15.259167 decode.go:60: Preambles: 111110010101001100000,0001011010100011
rtlamr2mqtt_1  | [2022-06-13 18:44:15] 18:44:15.259360 main.go:124: GainCount: 29
rtlamr2mqtt_1  | [2022-06-13 18:44:28] {"Time":"2022-06-13T18:44:28.281944012Z","Offset":0,"Length":0,"Type":"SCM","Message":{"ID":51643176,"Type":12,"TamperPhy":1,"TamperEnc":0,"Consumption":507681,"ChecksumVal":49096}}
rtlamr2mqtt_1  | [2022-06-13 18:44:28] Meter "51643176" - Consumption 005076.81. Sending value to MQTT.
rtlamr2mqtt_1  | [2022-06-13 18:44:28] Sending MQTT autodiscovery payload to Home Assistant...
rtlamr2mqtt_1  | [2022-06-13 18:44:28] Sending message to MQTT:
rtlamr2mqtt_1  | [2022-06-13 18:44:28]  > topic => homeassistant/sensor/rtlamr/gas_meter/config
rtlamr2mqtt_1  | [2022-06-13 18:44:28]  > payload => {"name": "gas_meter", "unique_id": "51643176", "unit_of_measurement": "CCF", "icon": "mdi:fire", "availability_topic": "rtlamr/status", "state_class": "total_increasing", "state_topic": "rtlamr/51643176/state", "json_attributes_topic": "rtlamr/51643176/attributes"}
rtlamr2mqtt_1  | [2022-06-13 18:44:28]  > qos => 1
rtlamr2mqtt_1  | [2022-06-13 18:44:28]  > retain => True
rtlamr2mqtt_1  | [2022-06-13 18:44:28] Unknown exception connecting to MQTT broker: Client.tls_set() got an unexpected keyword argument 'tls_insecure'
rtlamr2mqtt_1  | [2022-06-13 18:44:28] Sending message to MQTT:
rtlamr2mqtt_1  | [2022-06-13 18:44:28]  > topic => rtlamr/51643176/attributes
rtlamr2mqtt_1  | [2022-06-13 18:44:28]  > payload => {"Message Type": "SCM", "ID": 51643176, "Type": 12, "TamperPhy": 1, "TamperEnc": 0, "Consumption": 507681, "ChecksumVal": 49096}
rtlamr2mqtt_1  | [2022-06-13 18:44:28]  > retain => True
rtlamr2mqtt_1  | [2022-06-13 18:44:28] Unknown exception connecting to MQTT broker: Client.tls_set() got an unexpected keyword argument 'tls_insecure'
rtlamr2mqtt_1  | [2022-06-13 18:44:28] Sending message to MQTT:
rtlamr2mqtt_1  | [2022-06-13 18:44:28]  > topic => rtlamr/51643176/state
rtlamr2mqtt_1  | [2022-06-13 18:44:28]  > payload => 005076.81
rtlamr2mqtt_1  | [2022-06-13 18:44:28]  > retain => True
rtlamr2mqtt_1  | [2022-06-13 18:44:28] Unknown exception connecting to MQTT broker: Client.tls_set() got an unexpected keyword argument 'tls_insecure'
```

closed 2022-06-13 by bachya