allangood/rtlamr2mqtt #143: Add on stopping

markwoodward86, 2022-06-10:
Add on seems to just sort of stop. I will try and explain as best as i can. The logs do not show any errors, they simply start up as normal, grab one reading per meter defined (2 meters total for me) then sleep for the defined time... Then simply stop. The add on is still enabled. I rolled back a version or two and this continued. The issue started around 24 ish hours ago. 

I am typing on my cell here so formatting may be off. The logs below stop at 7:06 but the time now is 7:26. The difference is only 20 minutes in this case as i restarted the add on recently but this spanned to 7 hours overnight with no new lines of log. 

```
general:
  sleep_for: 300
  verbosity: debug
  listen_only: false
  tickle_rtl_tcp: true
  device_id: 0bda:2838
  usb_reset: '001:002'
mqtt:
  ha_autodiscovery: true
  ha_autodiscovery_topic: homeassistant
  base_topic: rtlamr
  tls_enabled: false
  tls_ca: /etc/ssl/certs/ca-certificates.crt
  tls_cert: /etc/ssl/my_self_signed_cert.crt
  tls_keyfile: /etc/ssl/my_self_signed_cert_key.key
  tls_insecure: true
  host: core-mosquitto
  port: 1883
  user: homeassistant
  password: *********
custom_parameters:
  rtltcp: '-s 2048000'
  rtlamr: '-unique=true'
meters:
  - id: 1552820016
    protocol: scm+
    name: production_meter
    format: '#####.##'
    unit_of_measurement: kWh
    icon: mdi:transmission-tower-export
  - id: 1552820019
    protocol: scm+
    name: consumption_meter
    format: '#####.##'
    unit_of_measurement: kWh
    icon: mdi:transmission-tower-import
```

[```
2022-06-10 06:57:54]  > username => homeassistant
[2022-06-10 06:57:54]  > client_id => rtlamr2mqtt
[2022-06-10 06:57:54]  > base_topic => rtlamr
[2022-06-10 06:57:54]  > availability_topic => rtlamr/status
[2022-06-10 06:57:54]  > tls => None
[2022-06-10 06:57:54] Reseting USB device: /dev/bus/usb/001/002
[2022-06-10 06:57:55] Reset sucessful.
[2022-06-10 06:57:55] Sending message to MQTT:
[2022-06-10 06:57:55]  > topic => rtlamr/status
[2022-06-10 06:57:55]  > payload => online
[2022-06-10 06:57:55]  > retain => True
[2022-06-10 06:57:55] Trying to start RTL_TCP: /usr/bin/rtl_tcp -d 0 -s 2048000
[2022-06-10 06:57:55] RTL_TCP started with PID 10
[2022-06-10 06:58:00] RTL_TCP is ready to receive connections!
[2022-06-10 06:58:00] Attempting to tune rtl_tcp to a different freq to shake things up
[2022-06-10 06:58:00] server: 127.0.0.1:1234, host: 127.0.0.1, port: 1234
[2022-06-10 06:58:00] Successfully tickled rtl_tcp
[2022-06-10 06:58:00] Trying to start RTLAMR: /usr/bin/rtlamr -msgtype=scm+,scm+ -format=json -filterid=1552820016,1552820019 -unique=true
[2022-06-10 06:58:00] RTLAMR started with PID 17
[2022-06-10 06:58:00] 06:58:00.491878 decode.go:45: CenterFreq: 912600155
[2022-06-10 06:58:00] 06:58:00.492280 decode.go:46: SampleRate: 2359296
[2022-06-10 06:58:00] 06:58:00.492305 decode.go:47: DataRate: 32768
[2022-06-10 06:58:00] 06:58:00.492317 decode.go:48: ChipLength: 72
[2022-06-10 06:58:00] 06:58:00.492327 decode.go:49: PreambleSymbols: 16
[2022-06-10 06:58:00] 06:58:00.492337 decode.go:50: PreambleLength: 2304
[2022-06-10 06:58:00] 06:58:00.492347 decode.go:51: PacketSymbols: 128
[... 50 lines cut ...]
[2022-06-10 07:01:50] Killed in the first attempt.
[2022-06-10 07:01:50] Sleeping for 300 seconds, see you later...
[2022-06-10 07:06:50] Reseting USB device: /dev/bus/usb/001/002
[2022-06-10 07:06:50] Reset sucessful.
[2022-06-10 07:06:50] Sending message to MQTT:
[2022-06-10 07:06:50]  > topic => rtlamr/status
[2022-06-10 07:06:50]  > payload => online
[2022-06-10 07:06:50]  > retain => True
[2022-06-10 07:06:51] Attempting to tune rtl_tcp to a different freq to shake things up
[2022-06-10 07:06:51] server: 127.0.0.1:1234, host: 127.0.0.1, port: 1234
[2022-06-10 07:06:51] Successfully tickled rtl_tcp
[2022-06-10 07:06:51] Trying to start RTLAMR: /usr/bin/rtlamr -msgtype=scm+,scm+ -format=json -filterid=1552820016,1552820019 -unique=true
[2022-06-10 07:06:51] RTLAMR started with PID 29
[2022-06-10 07:06:51] 07:06:51.902553 decode.go:45: CenterFreq: 912600155
[2022-06-10 07:06:51] 07:06:51.902754 decode.go:46: SampleRate: 2359296
[2022-06-10 07:06:51] 07:06:51.902765 decode.go:47: DataRate: 32768
[2022-06-10 07:06:51] 07:06:51.902772 decode.go:48: ChipLength: 72
[2022-06-10 07:06:51] 07:06:51.902777 decode.go:49: PreambleSymbols: 16
[2022-06-10 07:06:51] 07:06:51.902783 decode.go:50: PreambleLength: 2304
[2022-06-10 07:06:51] 07:06:51.902788 decode.go:51: PacketSymbols: 128
[2022-06-10 07:06:51] 07:06:51.902794 decode.go:52: PacketLength: 18432
[2022-06-10 07:06:51] 07:06:51.902801 decode.go:59: Protocols: scm+
[2022-06-10 07:06:51] 07:06:51.902807 decode.go:60: Preambles: 0001011010100011
[2022-06-10 07:06:51] 07:06:51.902813 main.go:124: GainCount: 29
```

markwoodward86, 2022-06-10:
As you can see, my entities are now at 35 minutes since updated

[image]

allangood, 2022-06-13:
It seems that you are running into this same problem #98 
Try to use a powered USB hub or a different USB device.

markwoodward86, 2022-06-13:
Understood, thanks Allan. This can be closed. I just restart the add on every 5 mins to workaround for now.

closed 2022-06-13 by markwoodward86