allangood/rtlamr2mqtt #144: Node found, debug seen, status = offline

dlasher, 2022-06-11:
Using built in HASS-addon (thank you) connected to remote node, successfully pulling lots of data. I can see ALL my neighbors. 

I've clearly identified which node is mine, and added it to the config. However, "rtlamr/status" always shows as offline, and nothing is appearing for useful data in HA.

Here's mine from rtlamr/debug:
`
{"Time": "2022-06-11T15:23:54.209700685-07:00", "Offset": 0, "Length": 0, "Type": "IDM", "Message": {"Preamble": 1431639715, "PacketTypeID": 28, "PacketLength": 92, "HammingCode": 198, "ApplicationVersion": 4, "ERTType": 7, "ERTSerialNumber": 26                                                          80xxxx, "ConsumptionIntervalCount": 74, "ModuleProgrammingState": 60, "TamperCounters": "AgEAGNkA", "AsynchronousCounters": 0, "PowerOutageFlags": "AAAAAAAA", "LastConsumptionCount": 5408106, "DifferentialConsumptionIntervals": [0, 0, 0, 0, 0, 0                                                          , 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0], "TransmitTimeOffset": 2387, "SerialNumberCRC": 31468, "PacketCRC": 42631}}
`

here's the config for meter:
`
- id: 2680xxxx
  protocol: idm
  name: House_Main
  format: '#####.##'
  unit_of_measurement: kWh
  icon: mdi:power-plug
  device_class: energy
`

the rtlamr2mqtt logs show data, but it doesn't appear to be going to anywhere but debug.
`
[2022-06-11 15:31:02] Sending message to MQTT:
[2022-06-11 15:31:02]  > topic => rtlamr/debug
[2022-06-11 15:31:02]  > payload => {"Time": "2022-06-11T15:31:02.409963438-07:00", "Offset": 0, "Length": 0, "Type": "IDM", "Message": {"Preamble": 1431639715, "PacketTypeID": 28, "PacketLength": 92, "HammingCode": 198, "ApplicationVersion": 4, "ERTType": 7, "ERTSerialNumber": 2680xxxx, "ConsumptionIntervalCount": 75, "ModuleProgrammingState": 60, "TamperCounters": "AgEAGNkA", "AsynchronousCounters": 0, "PowerOutageFlags": "AAAAAAAA", "LastConsumptionCount": 5408106, "DifferentialConsumptionIntervals": [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0], "TransmitTimeOffset": 4355, "SerialNumberCRC": 31468, "PacketCRC": 63266}}
[2022-06-11 15:31:02]  > retain => False
[2022-06-11 15:31:02] {"Time":"2022-06-11T15:31:02.832379161-07:00","Offset":0,"Length":0,"Type":"SCM","Message":{"ID":2680xxxx,"Type":7,"TamperPhy":2,"TamperEnc":1,"Consumption":5408106,"ChecksumVal":49581}}
`

What am I missing here?

allangood, 2022-06-11:
This listen_only mode wil never send anything to HA and its only purpose is
to help to find your meter id.

Once you have the id, you must set the mode to false.

dlasher, 2022-06-11:
\Facepalm

Giving that a shot - thank you.

dlasher, 2022-06-11:
Thank you. Working like a charm.

Given that I'm getting both IDM and SCM messages - which should I configure the meter to be?

{"name": "House_Main", "unique_id": "2680xxxx", "unit_of_measurement": "kWh", "icon": "mdi:power-plug", "availability_topic": "rtlamr/status", "state_class": "total_increasing", "state_topic": "rtlamr/2680xxxx/state", "json_attributes_topic": "rtlamr/2680xxxx/attributes", "device_class": "energy"}

allangood, 2022-06-12:
If both protocols work for your meter, then it really doesn't matter. You
can choose.

Regards

dlasher, 2022-06-12:
Thank you, apologies for bring up such an obvious issue..

closed 2022-06-12 by dlasher