allangood/rtlamr2mqtt #147: listen_only: true configuration option ignores rtlamr settings

BrettEBowman, 2022-06-18:
I am running rtlamr2mqtt version: 2.2.1. I am trying to change the centerfrequency and run in listen_only mode to see what meters I can find. However, whenever I enable listen_only mode, then the centerfrequency parameter is ignored.

For both of these tests, custom_parameters includes:
`rtlamr: '-unique=true -centerfreq=915000000 -symbollength=32 -tunergain=40'`

With listen_only=true, the logfile shows:
`[2022-06-18 11:34:22] Using "/data/options.json" config file
[2022-06-18 11:34:22] MQTT Host defined in config file. Ignoring Supervisor Configuration...
[2022-06-18 11:34:22] Starting in LISTEN ONLY Mode...
[2022-06-18 11:34:22] Using "/data/options.json" config file
[2022-06-18 11:34:22] MQTT Host defined in config file. Ignoring Supervisor Configuration...
[2022-06-18 11:34:22] Configured MQTT sender:
[2022-06-18 11:34:22]  > hostname => 192.168.7.16
[2022-06-18 11:34:22]  > port => 1883
[2022-06-18 11:34:22]  > username => mqtt-user
[2022-06-18 11:34:22]  > client_id => rtlamr2mqtt
[2022-06-18 11:34:22]  > base_topic => rtlamr
[2022-06-18 11:34:22]  > availability_topic => rtlamr/status
[2022-06-18 11:34:22]  > tls => None
[2022-06-18 11:34:22] Starting rtl_tcp with /usr/bin/rtl_tcp 
[2022-06-18 11:34:24] Starting rtlamr with ['/usr/bin/rtlamr', '-msgtype=all', '-format=json']
[2022-06-18 11:34:24] You should see all utility meters after this line:
[2022-06-18 11:34:24] 11:34:24.414671 decode.go:45: CenterFreq: 912600155
[2022-06-18 11:34:24] 11:34:24.415627 decode.go:46: SampleRate: 2359296
[2022-06-18 11:34:24] 11:34:24.415653 decode.go:47: DataRate: 32768
[2022-06-18 11:34:24] 11:34:24.415660 decode.go:48: ChipLength: 72
[2022-06-18 11:34:24] 11:34:24.415666 decode.go:49: PreambleSymbols: 32
[2022-06-18 11:34:24] 11:34:24.415673 decode.go:50: PreambleLength: 4608
[2022-06-18 11:34:24] 11:34:24.415678 decode.go:51: PacketSymbols: 736
[2022-06-18 11:34:24] 11:34:24.415684 decode.go:52: PacketLength: 105984
[2022-06-18 11:34:24] 11:34:24.415696 decode.go:59: Protocols: r900,scm,scm+,idm
[2022-06-18 11:34:24] 11:34:24.415706 decode.go:60: Preambles: 00000000000000001110010101100100,111110010101001100000,0001011010100011,01010101010101010001011010100011
[2022-06-18 11:34:24] 11:34:24.415718 main.go:124: GainCount: 29`

With listen_only: false, it shows that the parameters are being used:
`[2022-06-18 11:40:35] Using "/data/options.json" config file
[2022-06-18 11:40:35] MQTT Host defined in config file. Ignoring Supervisor Configuration...
[2022-06-18 11:40:35] RTLAMR2MQTT Starting...
[2022-06-18 11:40:35] RTL SDR Device 0bda:2838 found on USB port 001:004 - Index: 0
[2022-06-18 11:40:35] Configured MQTT sender:
[2022-06-18 11:40:35]  > hostname => 192.168.7.16
[2022-06-18 11:40:35]  > port => 1883
[2022-06-18 11:40:35]  > username => mqtt-user
[2022-06-18 11:40:35]  > client_id => rtlamr2mqtt
[2022-06-18 11:40:35]  > base_topic => rtlamr
[2022-06-18 11:40:35]  > availability_topic => rtlamr/status
[2022-06-18 11:40:35]  > tls => None
[2022-06-18 11:40:35] Reseting USB device: /dev/bus/usb/001/004
[2022-06-18 11:40:36] Reset sucessful.
[2022-06-18 11:40:36] Sending message to MQTT:
[2022-06-18 11:40:36]  > topic => rtlamr/status
[2022-06-18 11:40:36]  > payload => online
[2022-06-18 11:40:36]  > retain => True
[2022-06-18 11:40:36] Trying to start RTL_TCP: /usr/bin/rtl_tcp -d 0 -s 2048000
[2022-06-18 11:40:36] RTL_TCP started with PID 11
[2022-06-18 11:40:41] RTL_TCP is ready to receive connections!
[2022-06-18 11:40:41] Trying to start RTLAMR: /usr/bin/rtlamr -msgtype=scm+,scm+,scm -format=json -filterid=23382398,76622890,1234567 -unique=true -centerfreq=915000000 -symbollength=32 -tunergain=40
[2022-06-18 11:40:41] RTLAMR started with PID 16
[2022-06-18 11:40:41] 11:40:41.348532 decode.go:45: CenterFreq: 915000000
[2022-06-18 11:40:41] 11:40:41.349954 decode.go:46: SampleRate: 1048576
[2022-06-18 11:40:41] 11:40:41.349983 decode.go:47: DataRate: 32768
[2022-06-18 11:40:41] 11:40:41.349990 decode.go:48: ChipLength: 32
[2022-06-18 11:40:41] 11:40:41.349996 decode.go:49: PreambleSymbols: 21
[2022-06-18 11:40:41] 11:40:41.350002 decode.go:50: PreambleLength: 1344
[2022-06-18 11:40:41] 11:40:41.350008 decode.go:51: PacketSymbols: 128
[2022-06-18 11:40:41] 11:40:41.350014 decode.go:52: PacketLength: 8192
[2022-06-18 11:40:41] 11:40:41.350026 decode.go:59: Protocols: scm+,scm
[2022-06-18 11:40:41] 11:40:41.350034 decode.go:60: Preambles: 0001011010100011,111110010101001100000
[2022-06-18 11:40:41] 11:40:41.350045 main.go:124: GainCount: 29`

allangood, 2022-06-20:
Hi @BrettEBowman , this is due how the `listen_mode` was originally designed: to ignore the configuration file.
It is possible to pass these parameters using environment variables to the Docker container, but I agree it is not the best for the addon.

I am going to add this feature in the next minor release. :)

allangood, 2025-05-30:
I will close this as preparation for the next release.

closed 2025-05-30 by allangood