allangood/rtlamr2mqtt #152: Question about USB reset function

RMNB, 2022-07-20:
My SDR dongle stops responding about once per day. The only fix is to reset the USB port and restart the 
rtlamr2mqtt container. I've been using a shell script to do this but I just noticed the USB reset functionality.

If I add my dongle's USB id to the device_id option, will the USB port be reset after the sleep period?

allangood, 2022-07-20:
Yes, that is exactly what the USB reset function does :)

closed 2022-07-21 by allangood

RMNB, 2022-07-21:
Hi,

Not sure if you will still see this after the issue was closed but want to report something strange.

Likely unrelated but one of my meters (scm) stopped reporting after I added the device_id and restarted the container. The other meter (R900 water meter) was still working fine. IN LISTEN_ONLY mode both were reporting as usual. The solution was to change  to -symbollength=72 instead of 32 and now everything is working as usual.

allangood, 2022-08-08:
Yes, it is possible to see messages after we close the issue.

Thank you for the info.