allangood/rtlamr2mqtt #153: Is it possible to change the mhz of the frequency my meter runs at?

gbdavidx, 2022-07-23:
According to my gas meters website, P&E's gas SmartMeters™ operate in the 450-470 MHz frequency-band to communicate customer gas-usage to PG&E, how can i listen on the 460 Mhz band?

gbdavidx, 2022-07-23:
You will need to define 2 enviroment variables:

RTL_TCP_ARGS=nostart
RTLAMR_ARGS=-server=a.b.c.d:1234
If you are using the Add-on, then these parameters will be read from the configuration file.

Do you have a copy of what this would look like or a screenshot so I can see what it looks like and verify

arthurbenemann, 2022-08-05:
On my setup I have tweaked the default frequency  slightly to 916MHz. The meter I have (water r900) freq hops between 912 to 920, placing rtlamr in the middle of the band should increase the number of channels it can capture. I have not done extensive testing to confirm it, and also it does not matter since even though it transmits every 15s the data is only updated every 15mins.

To change the frequency center I added the `-centerfreq` to `custom_parameters` on the GUI config page. Currently it matches:
```
rtltcp: '-s 2048000'
rtlamr: '-unique=true -centerfreq=916000000'
```

Here is the log output when running at a different center frequency (edited meter ID):
```
[2022-08-05 15:19:44] Using "/data/options.json" config file
[2022-08-05 15:19:44] Fetching default MQTT configuration from http://supervisor/services/mqtt
[2022-08-05 15:19:44] RTLAMR2MQTT Starting...
[2022-08-05 15:19:44] RTL SDR Device 0bda:2838 found on USB port 001:003 - Index: 0
[2022-08-05 15:19:44] Configured MQTT sender:
[2022-08-05 15:19:44]  > hostname => core-mosquitto
[2022-08-05 15:19:44]  > port => 1883
[2022-08-05 15:19:44]  > username => addons
[2022-08-05 15:19:44]  > client_id => rtlamr2mqtt
[2022-08-05 15:19:44]  > base_topic => rtlamr
[2022-08-05 15:19:44]  > availability_topic => rtlamr/status
[2022-08-05 15:19:44]  > tls => None
[2022-08-05 15:19:44] Reseting USB device: /dev/bus/usb/001/003
[2022-08-05 15:19:45] Reset sucessful.
[2022-08-05 15:19:45] Sending message to MQTT:
[2022-08-05 15:19:45]  > topic => rtlamr/status
[2022-08-05 15:19:45]  > payload => online
[2022-08-05 15:19:45]  > retain => True
[2022-08-05 15:19:45] Trying to start RTL_TCP: /usr/bin/rtl_tcp -d 0 -s 2048000
[2022-08-05 15:19:45] RTL_TCP started with PID 10
[2022-08-05 15:19:50] RTL_TCP is ready to receive connections!
[2022-08-05 15:19:50] Trying to start RTLAMR: /usr/bin/rtlamr -msgtype=r900 -format=json -filterid=1234567890 -unique=true -centerfreq=916000000
[2022-08-05 15:19:50] RTLAMR started with PID 15
[2022-08-05 15:19:50] 15:19:50.281732 decode.go:45: CenterFreq: 916000000
[2022-08-05 15:19:50] 15:19:50.281822 decode.go:46: SampleRate: 2359296
[2022-08-05 15:19:50] 15:19:50.281827 decode.go:47: DataRate: 32768
[2022-08-05 15:19:50] 15:19:50.281834 decode.go:48: ChipLength: 72
[2022-08-05 15:19:50] 15:19:50.281836 decode.go:49: PreambleSymbols: 32
[2022-08-05 15:19:50] 15:19:50.281839 decode.go:50: PreambleLength: 4608
[2022-08-05 15:19:50] 15:19:50.281841 decode.go:51: PacketSymbols: 116
[2022-08-05 15:19:50] 15:19:50.281844 decode.go:52: PacketLength: 16704
[2022-08-05 15:19:50] 15:19:50.281847 decode.go:59: Protocols: r900
[2022-08-05 15:19:50] 15:19:50.281850 decode.go:60: Preambles: 00000000000000001110010101100100
[2022-08-05 15:19:50] 15:19:50.281853 main.go:124: GainCount: 29
[2022-08-05 15:19:59] {"Time":"2022-08-05T15:19:59.426173197-07:00","Offset":0,"Length":0,"Type":"R900","Message":{"ID":1234567890,"Unkn1":163,"NoUse":2,"BackFlow":0,"Consumption":15699504,"Unkn3":0,"Leak":0,"LeakNow":0}}
[2022-08-05 15:19:59] Meter "1234567890" - Consumption 1569950.4. Sending value to MQTT.
[2022-08-05 15:19:59] Sending MQTT autodiscovery payload to Home Assistant...
[2022-08-05 15:19:59] Sending message to MQTT:
[2022-08-05 15:19:59]  > topic => homeassistant/sensor/rtlamr/water_meter/config
[2022-08-05 15:19:59]  > payload => {"name": "water_meter", "unique_id": "1234567890", "unit_of_measurement": "gal", "icon": "mdi:water", "force_update": true, "availability_topic": "rtlamr/status", "state_class": "total_increasing", "state_topic": "rtlamr/1234567890/state", "json_attributes_topic": "rtlamr/1234567890/attributes", "device_class": "gas"}
[2022-08-05 15:19:59]  > qos => 1
[2022-08-05 15:19:59]  > retain => True
[2022-08-05 15:19:59] Sending message to MQTT:
[2022-08-05 15:19:59]  > topic => rtlamr/1234567890/attributes
[2022-08-05 15:19:59]  > payload => {"Message Type": "R900", "ID": 1234567890, "Unkn1": 163, "NoUse": 2, "BackFlow": 0, "Consumption": 15699504, "Unkn3": 0, "Leak": 0, "LeakNow": 0}
[2022-08-05 15:19:59]  > retain => True
[2022-08-05 15:19:59] Sending message to MQTT:
[2022-08-05 15:19:59]  > topic => rtlamr/1234567890/state
[2022-08-05 15:19:59]  > payload => 1569950.4
[2022-08-05 15:19:59]  > retain => True
[2022-08-05 15:19:59] Sleep_for defined, time to sleep!
[2022-08-05 15:19:59] Terminating all subprocess...
[2022-08-05 15:19:59] Kill process called.
[2022-08-05 15:19:59] Killing RTL_TCP...
[2022-08-05 15:20:04] Killed.
[2022-08-05 15:20:04] Killing RTLAMR...
[2022-08-05 15:20:04] Killed in the first attempt.
[2022-08-05 15:20:04] Sleeping for 300 seconds, see you later...
```

allangood, 2022-08-08:
I really not sure if it is possible to go to a so different frequency, but as mentioned by @arthurbenemann you can try using the `centerfreq`and `freqcorrection` to see if you can get the reading. But I don't think it will be a protocol that  RTLAMR will be able to decode. But, you can give it a try! :)

More here:
https://github.com/bemasher/rtlamr/wiki/Configuration

gbdavidx, 2022-08-09:
@arthurbenemann awesome thanks, how can i have it listen on all protocols?

[2022-08-08 18:27:18] Using "/data/options.json" config file
[2022-08-08 18:27:18] MQTT Host defined in config file. Ignoring Supervisor Configuration...
[2022-08-08 18:27:18] RTLAMR2MQTT Starting...
[2022-08-08 18:27:18] RTL SDR Device 0bda:2838 found on USB port 004:003 - Index: 0
[2022-08-08 18:27:18] No USB device specified in the config file, using the first found.
[2022-08-08 18:27:18] Configured MQTT sender:
[2022-08-08 18:27:18]  > hostname => 192.168.1.173
[2022-08-08 18:27:18]  > port => 1883
[2022-08-08 18:27:18]  > username => mqtt
[2022-08-08 18:27:18]  > client_id => rtlamr2mqtt
[2022-08-08 18:27:18]  > base_topic => rtlamr
[2022-08-08 18:27:18]  > availability_topic => rtlamr/status
[2022-08-08 18:27:18]  > tls => None
[2022-08-08 18:27:18] Reseting USB device: /dev/bus/usb/004/003
[2022-08-08 18:27:19] Reset sucessful.
[2022-08-08 18:27:19] Sending message to MQTT:
[2022-08-08 18:27:19]  > topic => rtlamr/status
[2022-08-08 18:27:19]  > payload => online
[2022-08-08 18:27:19]  > retain => True
[2022-08-08 18:27:19] Trying to start RTL_TCP: /usr/bin/rtl_tcp  -s 2048000
[2022-08-08 18:27:19] RTL_TCP started with PID 10
[2022-08-08 18:27:24] RTL_TCP is ready to receive connections!
[2022-08-08 18:27:24] Trying to start RTLAMR: /usr/bin/rtlamr -msgtype=scm+ -format=json -filterid=62331584 -unique=true -centerfreq=460000000
[2022-08-08 18:27:24] RTLAMR started with PID 15
[2022-08-08 18:27:24] 18:27:24.293076 decode.go:45: CenterFreq: 460000000
[2022-08-08 18:27:24] 18:27:24.294592 decode.go:46: SampleRate: 2359296
[2022-08-08 18:27:24] 18:27:24.294597 decode.go:47: DataRate: 32768
[2022-08-08 18:27:24] 18:27:24.294600 decode.go:48: ChipLength: 72
[2022-08-08 18:27:24] 18:27:24.294603 decode.go:49: PreambleSymbols: 16
[2022-08-08 18:27:24] 18:27:24.294605 decode.go:50: PreambleLength: 2304
[2022-08-08 18:27:24] 18:27:24.294608 decode.go:51: PacketSymbols: 128
[2022-08-08 18:27:24] 18:27:24.294610 decode.go:52: PacketLength: 18432
[2022-08-08 18:27:24] 18:27:24.294613 decode.go:59: Protocols: scm+
[2022-08-08 18:27:24] 18:27:24.294616 decode.go:60: Preambles: 0001011010100011
[2022-08-08 18:27:24] 18:27:24.294619 main.go:124: GainCount: 14

gbdavidx, 2022-08-09:
Weird, when i change it to Listen only true my center frequency went back to default..... seems a bug occured

rlust, 2022-08-11:
How do you set this to goto listen only?

allangood, 2022-11-05:
I will be closing as this is not a bug, but a question.
Please, use the Discussion board for questions.

Thank you

closed 2022-11-05 by allangood

valantur, 2023-05-05:
hey @gbdavidx were you ever able to figure this out? I'm a PG&E user as well and I don't know how to read the gas or electric meters.