allangood/rtlamr2mqtt #156: usb_reset resets the wrong device when not using device_id, only specifying a device with -d

ponchohoncho, 2022-08-28:
Hello! I'm trying to migrate my SDR devices to a raspberry pi and I'm having difficulty telling rtlamr to use a specific one. I'm trying to use both rtlamr and rtl433 with two identical sdr sticks. The device_id config won't work since I'm using two identical devices and they both have the same device id. I'm using the -d flag in the rtl_tcp command string in order to specify a certain stick, but since I'm not using the device_id config, rtlamr defaults to single, causing the below

```
[2022-08-28 15:46:40] RTLAMR2MQTT Starting...
[2022-08-28 15:46:40] Using "/etc/rtlamr2mqtt.yaml" config file
[2022-08-28 15:46:40] RTL SDR Device 0bda:2838 found on USB port 001:009 - Index: 0
[2022-08-28 15:46:40] RTL SDR Device 0bda:2838 found on USB port 001:004 - Index: 1
[2022-08-28 15:46:40] No USB device specified in the config file, using the first found.
[2022-08-28 15:46:40] Configured MQTT sender:
[2022-08-28 15:46:40]  > hostname => xxx
[2022-08-28 15:46:40]  > port => 1883
[2022-08-28 15:46:40]  > username => xxx
[2022-08-28 15:46:40]  > client_id => rtlamr2mqtt
[2022-08-28 15:46:40]  > base_topic => rtlamr
[2022-08-28 15:46:40]  > availability_topic => rtlamr/status
[2022-08-28 15:46:40]  > tls => None
[2022-08-28 15:46:40] Reseting USB device: /dev/bus/usb/001/004
[2022-08-28 15:46:40] Reset sucessful.
[2022-08-28 15:46:40] Sending message to MQTT:
[2022-08-28 15:46:40]  > topic => rtlamr/status
[2022-08-28 15:46:40]  > payload => online
[2022-08-28 15:46:40]  > retain => True
[2022-08-28 15:46:40] Trying to start RTL_TCP: /usr/bin/rtl_tcp  -d 0 -s 2048000
```

rtlamr finds 2 sticks, sees that device_id isn't set, so it says it's going to use the first one. i'd guess that the first one would be on port 001:009, index 0, but we see that it resets port 001:004, index 1. then it starts rtl_tcp with my -d 0 flag.

Would it be possible to have an supproted -d config in the yaml that would tell rtlamr which device index to use, so that it usb_resets that device instead of the "first found" one? I don't care which specific device it uses, I just need to tell it to use device index 0 so I can tell rtl433 to use device index 1. Right now rtlamr just resets the one that rtl433 is using, and I can't just swap them because it's not clear what the "first found" logic is like, so it might just swap back later.

allangood, 2022-08-28:
Thank you for your report.

Yes, this is something that requires a few changes in the id config.
I think it's better to switch to "usb port" instead of "device id".

This would make a break change unfortunately 😕, but better to address it
now than later.

Thank you!

ponchohoncho, 2022-08-28:
Unfortunately (for my use, anyways) defining a USB port would bring me back to square one since I can't define the same in rtl_433. Both containers let me use a -d flag, though.

ponchohoncho, 2022-08-28:
In the meantime, can we get a config option to disable usb_reset so that rtlamr doesn't reset another container's usb device?

allangood, 2022-08-28:
The "usb id" parameter creates a "-d" parameter when calling rtl_tcp. The
index is defined by the usb port sort order. This is how rtl_tcp does.

allangood, 2022-08-28:
Yes, let me see what I can do!

Thank you for your help!

allangood, 2022-09-15:
I've a change to the `device_id` parameter that will accept the USB port instead of the device id.
This way, the container will reset the usb device specified in that parameter.

The change has been added to the :dev tag and I will push to main after a few tests.

allangood, 2022-11-03:
The `device_id` parameter has been changed to support the USB ID and Port instead of device vendor. This will help with this bug.

In future releases, this will be the default behavior.

Fixed on #166

closed 2022-11-05 by allangood