allangood/rtlamr2mqtt #157: Python Error

mckellip, 2022-09-14:
I am getting an error when trying to start my docker. I tried doing a fresh install and get the same issue.

python3: error while loading shared libraries: libpython3.10.so.1.0: cannot open shared object file: No such file or directory

I am running on unraid version 6.10.3

allangood, 2022-09-14:
Could you give more info?
What docker command did you run? What is the full log message?

mckellip, 2022-09-14:
I am not sure I know how to get a lot of info because unraid uses a gui so I don't know a lot of the docker commands. But unraid did tell me it ran the command below. So when I go to command line and run it, it returns a sha output. Then I tried running the docker start command and it output the error.

```
# docker run -d -i --name='rtlamr2mqtt' --net='bridge' -e TZ="America/Denver" -e HOST_OS="Unraid" -e HOST_HOSTNAME="Tower" -e HOST_CONTAINERNAME="rtlamr2mqtt" -l net.unraid.docker.managed=dockerman -l net.unraid.docker.icon='/plugins/dynamix.docker.manager/images/question.png' -v '/mnt/user/applications/rtlamr2mqtt/':'/etc/':'rw' --device='/dev/bus/usb/005/004' 'allangood/rtlamr2mqtt'
fb49709acf7c961de8af051ec8c4d18e68f2344512cff9ae36abdcc3c8969c6b

# docker start -i rtlamr2mqtt
python3: error while loading shared libraries: libpython3.10.so.1.0: cannot open shared object file: No such file or directory
```

As for the full log message I don't know how to get more. I have tried looking in the docker logs too using this command.

```
# docker logs --details rtlamr2mqtt
python3: error while loading shared libraries: libpython3.10.so.1.0: cannot open shared object file: No such file or directory
```

allangood, 2022-09-15:
Duplicate #58 

This problem is due to the `-v '/mnt/user/applications/rtlamr2mqtt/':'/etc/':'rw'` that overrides the container `/etc`

closed 2022-09-15 by allangood

allangood, 2022-10-11:
Duplicated #58