allangood/rtlamr2mqtt #159: Supervisor Warning - Unknown option 'port'

stu1811, 2022-09-22:
Home Assistant 2022.9.6
Supervisor 2022.09.1
Frontend 20220907.2 - latest

`WARNING (MainThread) [supervisor.addons.options] Unknown option 'port' for rtlamr2mqtt (6713e36e_rtlamr2mqtt)`

sargonas, 2022-10-07:
Are you running on the default port? If so try removing the port line. I don't use it myself and am on default.. I did try adding the line to my config with the default port set and got the same error, fwiw.

sargonas, 2022-10-07:
Actually I think this might be because "port" is not declared inthe config.yaml that HA uses as a vaidator?

try this fork of mine: https://github.com/sargonas/rtlamr2mqtt
I added a port declaration to see if that makes a difference for you. I would test it locally but I am on a plane atm, but will try later tonight myself if I haven't heard back by then.

blandry3, 2022-10-07:
Similar issue here. I'm trying to use port 11298, but the log says 👍 

`[2022-10-07 16:29:01]  > port => 1883`

and

`[2022-10-07 16:29:06] Unknown exception connecting to MQTT broker: timed out`

stu1811, 2022-10-10:
> Actually I think this might be because "port" is not declared in the config.yaml that is used as a validator?
> 
> try this fork of mine: https://github.com/sargonas/rtlamr2mqtt (or change the config.yaml locally) it has one simply change of adding `port: str?` under `host: str?` in the rtlamr2mqtt-addon/config.yaml file
> 
> I would test it locally but I am on a plane atm, but will try later tonight myself if I haven't heard back by then.

Here is my mqtt config:
```
mqtt:
  ha_autodiscovery: true
  ha_autodiscovery_topic: homeassistant
  base_topic: rtlamr
  tls_enabled: false
  host: core-mosquitto
  port: 1883
  user: "!secrets mqtt_user"
  password: "!secrets mqtt_pass"
```

I tried modified `addons/git/6713e36e/rtlamr2mqtt-addon/config.yaml` per your suggestion. Restarting the addon and rebuilding and then restarting didn't stop the warnings in Supervisor. Do I need to do anything else to force the container to update?

The update...

```
  mqtt:
    host: str?
    port: str?
    user: str?
    password: str?
    ha_autodiscovery: bool?
    ha_autodiscovery_topic: str?
    base_topic: str
    tls_enabled: bool?
    tls_ca: str?
    tls_cert: str?
    tls_keyfile: str?
    tls_insecure: bool?
```

allangood, 2022-10-19:
I was away for a few weeks and couldn't follow this issue.
I will take a look and it as soon as possible.

Thank you for reporting.

allangood, 2022-11-04:
It should be fixed now.

stu1811, 2022-11-04:
New version gets a different error:
```
[2022-11-03 20:56:28] Using "/data/options.json" config file
[2022-11-03 20:56:28] MQTT Host defined in config file. Ignoring Supervisor Configuration...
[2022-11-03 20:56:28] RTLAMR2MQTT Starting...
[2022-11-03 20:56:28] RTL SDR Device 0bda:2832 found on USB port 005:002 - Index: 0
Traceback (most recent call last):
  File "/usr/bin/rtlamr2mqtt.py", line 440, in <module>
    elif re.match(r"(^[0-9]{3}:([0-9]{3}$)))", usb_device_id) is not None:
  File "/usr/local/lib/python3.10/re.py", line 190, in match
    return _compile(pattern, flags).match(string)
  File "/usr/local/lib/python3.10/re.py", line 303, in _compile
    p = sre_compile.compile(pattern, flags)
  File "/usr/local/lib/python3.10/sre_compile.py", line 788, in compile
    p = sre_parse.parse(p, flags)
  File "/usr/local/lib/python3.10/sre_parse.py", line 969, in parse
    raise source.error("unbalanced parenthesis")
re.error: unbalanced parenthesis at position 23
```

allangood, 2022-11-05:
#174 fix this issue

closed 2022-11-06 by allangood