allangood/rtlamr2mqtt #165: Add support for device_class: water (HA 2022.11)

jcollas, 2022-11-02:
I think the config.yaml needs to be extended to support a device_class of water, per the latest energy feature in HA 2022.11

bartekd123, 2022-11-03:
I think it is there. I set the following in my config.yaml, and I see the device_class being sent as water, but it doesn't show up as a water source, so don't know whats up. If you figure out anything let me know.

```
    # A device_class to define what the sensor is measuring for use in the Energy panel
    # Typical values are "gas" or "energy". Default is blank.
    device_class: "water"
    # "total_increasing" for most meters, "total" for meters that might go
    # backwards (net energy meters). Defaults to "total_increasing" if unset.
    # state_class:

```

 When I look in the state in homeassistant, I can see it being sent:

```
state_class: total_increasing
unit_of_measurement: m³
device_class: water
icon: mdi:gauge
friendly_name: meter_water
Message Type: SCM+
FrameSync: 5795
ProtocolID: 30
EndpointType: 171
EndpointID: <redacted>
Consumption: 276634
Tamper:  <redacted>
PacketCRC: <redacted>

```

jcollas, 2022-11-03:
I'm running 2.2.1, and when I set the device_class to 'water' in the config, I get this error:

Failed to save add-on configuration, value must be one of ['aqi', 'battery', 'carbon_dioxide', 'carbon_monoxide', 'current', 'date', 'energy', 'frequency', 'gas', 'humidity', 'illuminance', 'monetary', 'nitrogen_dioxide', 'nitrogen_monoxide', 'nitrous_oxide', 'none', 'ozone', 'pm1', 'pm10', 'pm25', 'power', 'power_factor', 'pressure', 'signal_strength', 'sulphur_dioxide', 'temperature', 'timestamp', 'volatile_organic_compounds', 'voltag'].

Are you making your change in the add-on UI?

bartekd123, 2022-11-03:
I did it right in the yaml. There are comments in the yaml for device_class, so maybe try updating and putting device class in the meter section like I did? Not sure why it sends for me. Even though it doesn’t work for the energy because the statistics are not working for that value.

bartekd123, 2022-11-03:
I just looked at the one example yaml and at the bottom there is a list. Maybe you need to add water in there too.

allangood, 2022-11-03:
Yes @jcollas , you are correct. The config doesn't have all supported device_classes by the latest HA version (2022.11).

#166 fix this.

arthurbenemann, 2022-11-04:
From https://github.com/allangood/rtlamr2mqtt/issues/165#issuecomment-1302379824 it looks like it has been fixed. I have not updated as I'm getting a `usb` device error with the new version.

In the meantime I have been using the following to be able to add my (`gallon` counting) meter to the energy dashboard:
```yaml
template:
  - sensor:
      - name: water_meter_m3
        unit_of_measurement: "m³"
        device_class: water
        state_class: total_increasing
        state: >
          {{ (states('sensor.water_meter')|float)/264.172}}
 ```

arthurbenemann, 2022-11-04:
Thanks for fixing both #168 and #165 @allangood! Updated to ` 2.3.3 ` and everything works out of the box now, including adding it to the energy dashboard.

For my r900 meter  (id obfuscated) I used the following config on the meters section:
```yaml
- id: 18xxxxxxxx
  protocol: r900
  name: water_meter
  format: "#######.#"
  unit_of_measurement: gal
  icon: mdi:water
  device_class: water

```

closed 2022-11-05 by allangood