allangood/rtlamr2mqtt #168: NameError: name 'usb_port' is not defined

socbrian, 2022-11-03:
Been getting some weird issues starting the add-on. I defined my device ID device_id: 0bda:2838 in the config, but been getting this: 

`Traceback (most recent call last):
  File "/usr/bin/rtlamr2mqtt.py", line 499, in <module>
    reset_usb_device(usb_port)
NameError: name 'usb_port' is not defined`

The add-on finds the radio too:

`[2022-11-03 14:26:22] RTL SDR Device 0bda:2838 found on USB port 001:006 - Index: 0`

If I just leave the device ID as single, I get a different issue: 

`[2022-11-03 14:30:47] RTL SDR Device 0bda:2838 found on USB port 001:006 - Index: 0
Traceback (most recent call last):
  File "/usr/bin/rtlamr2mqtt.py", line 439, in <module>
    elif re.match(r"(^[0-9]{3}:([0-9]{3}$)))", usb_device_id) is not None:
  File "/usr/local/lib/python3.10/re.py", line 190, in match
    return _compile(pattern, flags).match(string)
  File "/usr/local/lib/python3.10/re.py", line 303, in _compile
    p = sre_compile.compile(pattern, flags)
  File "/usr/local/lib/python3.10/sre_compile.py", line 788, in compile
    p = sre_parse.parse(p, flags)
  File "/usr/local/lib/python3.10/sre_parse.py", line 969, in parse
    raise source.error("unbalanced parenthesis")
re.error: unbalanced parenthesis at position 23`

afly007, 2022-11-03:
I'm having the same issue, I just installed this today after upgrading to 2022.11.1.

bartekd123, 2022-11-03:
Yea, I am getting that too. @allangood  do you know what could have caused this in the last update?

allangood, 2022-11-04:
Looks like I've pushed a change that I was still working on it :(
Sorry about that.

I will be working on fixing this ASAP and will update here when the new PR came through.

allangood, 2022-11-04:
It should be fixed now. Sorry about that :(
PR #170

bartekd123, 2022-11-04:
trying to update, but its not pulling the latest image. Itll take time before its updated for allangood/rtlamr2mqtt:main, or should I use a tag other than main?

socbrian, 2022-11-04:
> It should be fixed now. Sorry about that :( PR #170

Works for me! Thanks for all the hard work. I even got the water meter on the new energy dashboard!!

closed 2022-11-04 by socbrian