allangood/rtlamr2mqtt #169: Integrating American Meter Co. model AC-250 in HA with this Add-on

moasson, 2022-11-03:
Hi all,

First of all, thanks for this project, I was able to monitor my electricity using the rtl-sdr dongle and this HA add-on. Very cool.

This led me to try to integrate my gas meter consumption into HA as well, however, I am unable to locate my meter when running the add-on in docker using the DEBUG MODE.

My gas meter is model AC-250 from American Meter Company. Hope someone here can assist me on making this work if it possible.

[image]

allangood, 2022-11-04:
Sorry, but I don't work in the "supporting devices" itself.
In a nutshell, this project only wraps the RTLAMR into a Docker container to ship the output in a way HA can understand.

The right place to ask for new devices would be here:
https://github.com/bemasher/rtlamr

Regards

closed 2022-11-04 by allangood

moasson, 2022-11-04:
Will do, thank you!