allangood/rtlamr2mqtt #173: Latest update doesnt allow add on to start

markwoodward86, 2022-11-04:
Error logs show:
```

[2022-11-04 07:46:19] Using "/data/options.json" config file
[2022-11-04 07:46:19] MQTT Host defined in config file. Ignoring Supervisor Configuration...
[2022-11-04 07:46:19] RTLAMR2MQTT Starting...
[2022-11-04 07:46:19] RTL SDR Device 0bda:2838 found on USB port 001:003 - Index: 0
Traceback (most recent call last):
  File "/usr/bin/rtlamr2mqtt.py", line 440, in <module>
    elif re.match(r"(^[0-9]{3}:([0-9]{3}$)))", usb_device_id) is not None:
  File "/usr/local/lib/python3.10/re.py", line 190, in match
    return _compile(pattern, flags).match(string)
  File "/usr/local/lib/python3.10/re.py", line 303, in _compile
    p = sre_compile.compile(pattern, flags)
  File "/usr/local/lib/python3.10/sre_compile.py", line 788, in compile
    p = sre_parse.parse(p, flags)
  File "/usr/local/lib/python3.10/sre_parse.py", line 969, in parse
    raise source.error("unbalanced parenthesis")
re.error: unbalanced parenthesis at position 23
```

markwoodward86, 2022-11-04:
I rolled back to 2.2.1 and no issues. Looks like the issue may be with my local python though, thoughts?

taylorcoffelt, 2022-11-04:
Getting this as well with 2.3.3, the issue looks like an extra parentheses at the end of the regex on line 440:
 https://github.com/allangood/rtlamr2mqtt/blob/main/rtlamr2mqtt-addon/rtlamr2mqtt.py#L440

taylorcoffelt, 2022-11-04:
@markwoodward86 Add-ons run in separate containers, so your local python wouldn't affect what's running for this add-on, since it uses it's own shipped version from the container [(python:3.10)](https://github.com/allangood/rtlamr2mqtt/blob/main/rtlamr2mqtt-addon/Dockerfile#L15)

markwoodward86, 2022-11-04:
Thanks for the clarification! I removed my comment to avoid confusion.

taylorcoffelt, 2022-11-04:
I've found a workaround until the regex is fixed.

Run `lsusb` in the terminal and check the IDs listed against the ones in https://github.com/allangood/rtlamr2mqtt/blob/main/rtlamr2mqtt-addon/sdl_ids.txt

Once you've found a match, you can use that in your `device_id` for the configuration instead of `single`, and prevent the script from hitting the bad regex.

taylorcoffelt, 2022-11-04:
I've created PR https://github.com/allangood/rtlamr2mqtt/pull/174 with the needed changes

allangood, 2022-11-05:
#174 has been merged.
Thank you @taylorcoffelt

karepiu, 2022-11-06:
Any ETA for release of fixed version ?

allangood, 2022-11-06:
@karepiu the latest version had reports to be fixed.
Are you using the latest 2.3.3 version?

karepiu, 2022-11-07:
@allangood yes i am using 2.3.3. This is the version this issue has show up with (in my case).

SpaceCadetRick, 2022-11-08:
I'm also on 2.3.3 and having the same issue, add-on won't start.

taylorcoffelt, 2022-11-08:
[image]

Have you tried rebuilding the add-on? (Click "rebuild" and start the add-on after the rebuild is complete)
It's possible your local container version is out-of-date, even if you have the latest 2.3.3.

SpaceCadetRick, 2022-11-08:
I had done it before but just went ahead and hit rebuild again and then hit start once it finished and it still doesn't work.

allangood, 2022-11-08:
Any console output? Debug messages?
Sorry but "it doesn´t start" doesn´t help much to find the problem.

SpaceCadetRick, 2022-11-08:
I get the following error message after starting the add-on and trying to view the logs. 

2022-11-08 15:02:34.576 ERROR (MainThread) [homeassistant.components.hassio] Failed to to call /addons/6713e36e_rtlamr2mqtt/stats - Can't read stats from addon_6713e36e_rtlamr2mqtt: Expecting value: line 1 column 1 (char 0)

allangood, 2022-11-08:
@SpaceCadetRick this is completely unrelated with this issue.
Try to save your config and delete/re-install or rebuild the addon. This is something local to your installation

closed 2022-11-11 by allangood