allangood/rtlamr2mqtt #178: Allow Meter ID to be stored in secrets.yaml

ayourk, 2022-11-15:
It would be great to be able to do something like the following in secrets,yaml and then use that in the config to hide the meter IDs in the main config.

--- secrets.yaml ---
meter_elec: 10528282
meter_gas: 52212210
meter_water: 3223125100
--- --- --- --- ---

The numbers above are random numbers I created in the moment and are not reflective of an exact meter ID; they are used as examples only.
```--- meters config ---
- id: !secret meter_elec 
  protocol: scm
  name: meter_electricity
  format: "#####.##"
  unit_of_measurement: kWh
  icon: mdi:lightning-bolt
  device_class: energy
  state_class: total_increasing
  filters:
    - multiply: 0.01
- id: !secret meter_gas
  protocol: scm
  name: meter_gas
  format: "####.###"
  unit_of_measurement: ft³
  icon: mdi:gas-meter
  device_class: gas
  state_class: total_increasing
  filters:
    - multiply: 0.001
- id: !secret meter_water
  protocol: r900
  name: meter_water
  format: "#######.#"
  unit_of_measurement: gal
  icon: mdi:water
  device_class: water
  state_class: total_increasing
  filters:
    - multiply: 0.1
```
--- --- --- --- --- ---

Please note:  If you want to add Gas or Water to the Energy Dashboard, you MUST change the units to m³.  Once added to the energy dashboard, the units can be changed back.

onedr0p, 2023-12-18:
Another improvement to this would to also be able to support injecting env like this

```yaml
- id: !env METER_ID
  protocol: r900
  name: meter_water
  format: "#######.#"
  unit_of_measurement: gal
  icon: mdi:water
  device_class: water
  state_class: total_increasing
  filters:
    - multiply: 0.1
```

PrplHaz4, 2023-12-18:
Either of these methods for pulling Meter IDs out of the primary config would work for me.
> 
> Please note: If you want to add Gas or Water to the Energy Dashboard, you MUST change the units to m³. Once added to the energy dashboard, the units can be changed back.

Can you elaborate on this one?  I've been wondering why my water units aren't showing up in the Energy Dashboard - they are currently in ft³.  Something like this?
1. Change the units to m³ in rtlamr2mqtt, restart rtlamr2mqtt
2. Restart homeassistant, see them on the dashboard
3. Reset units back to ft³ in rtlamr2mqtt, restart rtlamr2mqtt
4. Restart homeassistant, see them on the dashboard

ayourk, 2023-12-18:
> Either of these methods for pulling Meter IDs out of the primary config would work for me.
>
>
> Can you elaborate on this one? I've been wondering why my water units aren't showing up in the Energy [...]

I don't believe I even had to restart Home Assistant.  I restarted rtlamr2mqtt so many times during the process of getting my config setup that I don't remember if this is required.