allangood/rtlamr2mqtt #179: Loss of ha_autodiscovery in MQTT

mkmer, 2022-11-17:
Some event occurs - maybe MQTT reset?, where the ha_autodiscovery information is no longer published on the MQTT server.  The rest of the application is running and updating meter data, but HA "says" the entity is no longer being provided.

Easiest fix would be to send the ha_autodiscovery MQTT every time.  A bit more elegant maybe to look up the ha_autodiscovery data each time data is published, and if it's missing resend it.

Restarting the container/app always "fixes" HA.

Ramias1, 2022-11-21:
I just noticed this today too.  Restarted all of my containers.  Now the electricity meter still shows in MQTT but not the gas meter.

allangood, 2022-11-21:
The HA autodiscovery paylod is sent using MQTT `retain=True`. This means it will survive and persists after been read by HA.

This behavior is expected only if your MQTT broker loses its database when recreated/restarted.
How your MQTT broker is configured? It has a persistent data/volume for the database?

Ramias1, 2022-11-21:
My broker is not persistent.  I had to delete my stacks in Portainer and when it redeployed it probably got a new volume.  This makes sense.  Thanks

mkmer, 2022-11-21:
Mine is set to persist and MQTT explorer indicates it is "retained".  Not sure how it disappeared.  Could be more robust if it checked and published if missing.

On further research, the missing discovery data may not be the only issue causing unavailability.  There are times where the rtlamr/status = "offline".  This may be part of the problem.  Checking the container logs, the app is running, and the log indicate successful publishing of data to the MQTT server.  What sets that tag true?

allangood, 2022-11-21:
Having to check for a message will require more resources to keep checking it. The way this project works is to send an `autodiscovery` message every time a reading happens.

The `status` payload is set by the LWT configuration, i.e. by your MQTT broker.
Double check your MQTT broker and the communication between rtlamr2mqtt and your broker.

closed 2022-11-21 by allangood