allangood/rtlamr2mqtt #181: Listen Only Mode Only Works For 6 Mins

ADeadPixel, 2022-11-19:
I'm currently trying to figure out my meters information so I am running this add-on in listen only mode and monitoring the debug topic in HA and in MQTT Explorer. However, after 6 mins of running the receiver context will cancel and the i/o will time out.

`[2022-11-18 18:55:03] Using "/data/options.json" config file
[2022-11-18 18:55:03] MQTT Host defined in config file. Ignoring Supervisor Configuration...
[2022-11-18 18:55:03] Starting in LISTEN ONLY Mode...
[2022-11-18 18:55:03] Using "/data/options.json" config file
[2022-11-18 18:55:03] MQTT Host defined in config file. Ignoring Supervisor Configuration...
[2022-11-18 18:55:03] Configured MQTT sender:
[2022-11-18 18:55:03]  > hostname => --------
[2022-11-18 18:55:03]  > port => --------
[2022-11-18 18:55:03]  > username => --------
[2022-11-18 18:55:03]  > client_id => rtlamr2mqtt
[2022-11-18 18:55:03]  > base_topic => rtlamr
[2022-11-18 18:55:03]  > availability_topic => rtlamr/status
[2022-11-18 18:55:03]  > tls => None
[2022-11-18 18:55:03] Starting rtl_tcp with /usr/bin/rtl_tcp 
[2022-11-18 18:55:05] Starting rtlamr with ['/usr/bin/rtlamr', '-msgtype=all', '-format=json']
[2022-11-18 18:55:05] You should see all utility meters after this line:
[2022-11-18 18:55:05] 18:55:05.454244 decode.go:45: CenterFreq: 912600155
[2022-11-18 18:55:05] 18:55:05.454534 decode.go:46: SampleRate: 2359296
[2022-11-18 18:55:05] 18:55:05.454547 decode.go:47: DataRate: 32768
[2022-11-18 18:55:05] 18:55:05.454556 decode.go:48: ChipLength: 72
[2022-11-18 18:55:05] 18:55:05.454564 decode.go:49: PreambleSymbols: 32
[2022-11-18 18:55:05] 18:55:05.454577 decode.go:50: PreambleLength: 4608
[2022-11-18 18:55:05] 18:55:05.454585 decode.go:51: PacketSymbols: 736
[2022-11-18 18:55:05] 18:55:05.454592 decode.go:52: PacketLength: 105984
[2022-11-18 18:55:05] 18:55:05.454604 decode.go:59: Protocols: scm+,idm,r900,scm
[2022-11-18 18:55:05] 18:55:05.454613 decode.go:60: Preambles: 0001011010100011,01010101010101010001011010100011,00000000000000001110010101100100,111110010101001100000
[2022-11-18 18:55:05] 18:55:05.454637 main.go:124: GainCount: 29
[2022-11-18 19:01:45] 19:01:45.447843 main.go:343: Receiver context cancelled.
[2022-11-18 19:01:45] 19:01:45.448288 main.go:320: read tcp 127.0.0.1:40528->127.0.0.1:1234: i/o timeout
[2022-11-18 19:01:45] io.ReadFull
[2022-11-18 19:01:45] main.(*Receiver).Run.func2
[2022-11-18 19:01:45] 	/go/pkg/mod/github.com/bemasher/rtlamr@v0.9.3/main.go:181
[2022-11-18 19:01:45] runtime.goexit
[2022-11-18 19:01:45] 	/usr/local/go/src/runtime/asm_amd64.s:1594`

I'm not sure if listen_only mode is meant to run for more than 6 mins. But idk how else to get my meter info. I did not pick up any info during those 6 mins. That makes me think that the meters either trigger at a certain times of day or when the trucks roll by they trigger it somehow. Either way I would need to be in listen_only mode to see if I can scrape the data when/if that happens.

Slightly outside of the scope of this issue I did find my electric meter in bemasher's csv but I'm not sure where to find the meter id information on it. FCC ID: R7PEG1R1S6 if anyone wants to assist.

allangood, 2022-11-21:
This seems to be a hardware issue.
Try to plug your SDL device on an powered USB hub.

closed 2022-11-21 by allangood

allangood, 2022-11-21:
Please, check #98 for reference

ADeadPixel, 2022-11-27:
I have tried using a powered USB hub as mentioned in #98 but it did not solve the issue. I have a very similar setup to kspearrin where my Home Assistant is a VM inside of a proxmox server. I'm going to try this again on one of my pis and see if anything changes.

Sorry for late response (holidays and what not)

ADeadPixel, 2022-11-27:
I am experiencing the same issue on a fresh pi with a fresh docker image. See logs below

```
------@rtlsdr:~$ sudo docker run --rm -ti -e LISTEN_ONLY=yes -e RTL_MSGTYPE="all" --device=/dev/bus/usb:/dev/bus/usb allangood/rtlamr2mqtt
[2022-11-27 16:47:13] Starting in LISTEN ONLY Mode...
[2022-11-27 16:47:13] No Supervisor detected.
[2022-11-27 16:47:13] Starting rtl_tcp with /usr/bin/rtl_tcp
[2022-11-27 16:47:15] Starting rtlamr with ['/usr/bin/rtlamr', '-msgtype=all', '-format=json']
[2022-11-27 16:47:15] You should see all utility meters after this line:
[2022-11-27 16:47:15] 16:47:15.324168 decode.go:45: CenterFreq: 912380000
[2022-11-27 16:47:15] 16:47:15.324588 decode.go:46: SampleRate: 2359296
[2022-11-27 16:47:15] 16:47:15.324605 decode.go:47: DataRate: 32768
[2022-11-27 16:47:15] 16:47:15.324617 decode.go:48: ChipLength: 72
[2022-11-27 16:47:15] 16:47:15.324637 decode.go:49: PreambleSymbols: 32
[2022-11-27 16:47:15] 16:47:15.324648 decode.go:50: PreambleLength: 4608
[2022-11-27 16:47:15] 16:47:15.324658 decode.go:51: PacketSymbols: 736
[2022-11-27 16:47:15] 16:47:15.324668 decode.go:52: PacketLength: 105984
[2022-11-27 16:47:15] 16:47:15.324688 decode.go:59: Protocols: scm,scm+,idm,r900
[2022-11-27 16:47:15] 16:47:15.324713 decode.go:60: Preambles: 111110010101001100000,0001011010100011,01010101010101010001011010100011,00000000000000001110010101100100
[2022-11-27 16:47:15] 16:47:15.324738 main.go:124: GainCount: 29
[2022-11-27 16:54:45] 16:54:45.053523 main.go:343: Receiver context cancelled.
[2022-11-27 16:54:45] 16:54:45.054098 main.go:320: read tcp 127.0.0.1:60936->127.0.0.1:1234: i/o timeout
[2022-11-27 16:54:45] io.ReadFull
[2022-11-27 16:54:45] main.(*Receiver).Run.func2
[2022-11-27 16:54:45]   /go/pkg/mod/github.com/bemasher/rtlamr@v0.9.3/main.go:181
[2022-11-27 16:54:45] runtime.goexit
[2022-11-27 16:54:45]   /usr/local/go/src/runtime/asm_arm64.s:1172

```

Doing some more digging on this I found this exact issue [here](https://github.com/jdeath/RTLAMR2MQQT/issues/10). Looking into it I may have lost the lottery and received a bad unit. Going to return and see if a new one works better. Will update once new one comes in and I get a chance to test it out.

ADeadPixel, 2022-12-01:
A new antenna did no solve the issue. I'm going to try a different RTLamr project to see if I can get that one working. I'm not sure what the issue here could be.

MattInternet, 2023-02-08:
I also have a similar issue, running in a container in k3s. I got a powered USB hub today and am still seeing the same behavior.

```
2023-02-08 04:25:21.163064+00:00[2023-02-07 21:25:21] Trying to start RTLAMR: /usr/bin/rtlamr -msgtype=scm,scm+ -format=json -filterid=55755533,94975255 -unique=true -symbollength=32
2023-02-08 04:25:21.163393+00:00[2023-02-07 21:25:21] RTLAMR started with PID 1454
2023-02-08 04:25:21.168098+00:00[2023-02-07 21:25:21] 21:25:21.164902 decode.go:45: CenterFreq: 912600155
2023-02-08 04:25:21.168113+00:00[2023-02-07 21:25:21] 21:25:21.164979 decode.go:46: SampleRate: 1048576
2023-02-08 04:25:21.168118+00:00[2023-02-07 21:25:21] 21:25:21.164982 decode.go:47: DataRate: 32768
2023-02-08 04:25:21.168123+00:00[2023-02-07 21:25:21] 21:25:21.164984 decode.go:48: ChipLength: 32
2023-02-08 04:25:21.168128+00:00[2023-02-07 21:25:21] 21:25:21.164985 decode.go:49: PreambleSymbols: 21
2023-02-08 04:25:21.168132+00:00[2023-02-07 21:25:21] 21:25:21.164987 decode.go:50: PreambleLength: 1344
2023-02-08 04:25:21.168137+00:00[2023-02-07 21:25:21] 21:25:21.164989 decode.go:51: PacketSymbols: 128
2023-02-08 04:25:21.168142+00:00[2023-02-07 21:25:21] 21:25:21.164990 decode.go:52: PacketLength: 8192
2023-02-08 04:25:21.168146+00:00[2023-02-07 21:25:21] 21:25:21.164993 decode.go:59: Protocols: scm,scm+
2023-02-08 04:25:21.168151+00:00[2023-02-07 21:25:21] 21:25:21.164995 decode.go:60: Preambles: 111110010101001100000,0001011010100011
2023-02-08 04:25:21.168155+00:00[2023-02-07 21:25:21] 21:25:21.164997 main.go:124: GainCount: 29
2023-02-08 04:25:26.181196+00:00[2023-02-07 21:25:26] 21:25:26.170844 main.go:343: Receiver context cancelled.
2023-02-08 04:25:26.181276+00:00[2023-02-07 21:25:26] 21:25:26.171096 main.go:320: read tcp 127.0.0.1:42626->127.0.0.1:1234: i/o timeout
2023-02-08 04:25:26.181284+00:00[2023-02-07 21:25:26] io.ReadFull
2023-02-08 04:25:26.181290+00:00[2023-02-07 21:25:26] main.(*Receiver).Run.func2
2023-02-08 04:25:26.181295+00:00[2023-02-07 21:25:26] 	/go/pkg/mod/github.com/bemasher/rtlamr@v0.9.3/main.go:181
2023-02-08 04:25:26.181301+00:00[2023-02-07 21:25:26] runtime.goexit
2023-02-08 04:25:26.181307+00:00[2023-02-07 21:25:26] 	/usr/local/go/src/runtime/asm_amd64.s:1598
2023-02-08 04:25:26.181312+00:00[2023-02-07 21:25:26] Sleep_for defined, time to sleep!
```

mwatz1234, 2023-02-15:
I am seeing the same issue
```
[2023-02-12 18:17:01] 18:17:01.513155 main.go:343: Receiver context cancelled.
[2023-02-12 18:17:01] 18:17:01.513500 main.go:320: read tcp 127.0.0.1:47122->127.0.0.1:1234: i/o timeout
[2023-02-12 18:17:01] io.ReadFull
[2023-02-12 18:17:01] main.(*Receiver).Run.func2
[2023-02-12 18:17:01] 	/go/pkg/mod/github.com/bemasher/rtlamr@v0.9.3/main.go:181
[2023-02-12 18:17:01] runtime.goexit
[2023-02-12 18:17:01] 	/usr/local/go/src/runtime/asm_arm64.s:1172
```
I have tried a powered USB hub, having an usb extension so the usb hub won't get heated.  I have tried it on  and RTL-SDR Blog V3 R860 RTL2832U and Nooelec Dual-Band NESDR Nano 2+.

If I run the docker with filtering the meter's I have seen listed in the listen only mode, it will run without issues (likely because its sleeping after a successful read)