allangood/rtlamr2mqtt #185: MQTT LWT (last will and testament) doesn't work

melyux, 2023-01-01:
I found that if I ungracefully stop rtlamr2mqtt somehow (like pulling the plug or disconnecting the internet), the status will remain "online". It only becomes "offline" if rtlamr2mqtt gracefully shuts down. I noticed this when my rtlamr2mqtt went down for days but never showed an indication (especially insidious because rtlamr2mqtt publishes asynchronously, so right now there is no way to ever know whether rtlamr2mqtt is down or no gas has been used since the last MQTT update).

Not sure from a look at the code why the will doesn't work. But I've never used the `publish.single()` method of setting the will with Paho, I just use the `client.will_set()` in my projects, so I'm not experienced with the method used here.

allangood, 2023-01-10:
This project doesn´t keep a permanent connection with MQTT. This means that the LWT will only be set to "offline" if the container dies during a status update. If the container dies during a sleep time, then the only way to know if it is working is using the "last_updated" attribute from the sensor.

melyux, 2023-01-10:
That's the point of the LWT functionality in MQTT. It automatically sets the status to "offline" if your application loses connection to MQTT for longer than a set amount of time. It's not just for consciously setting it to "offline" manually.

allangood, 2023-01-11:
If you read my comment, you will see that this project will *not* keep a constant connection with MQTT.
It connects, send the payload and then disconnects. Keeping an constant connection is out of scope for this project as it is possible to catch this using the `last_updated` attribute in HA.

melyux, 2023-01-11:
In that case the status topic is misleading. Why have it at all if it's purposefully not reliable?

melyux, 2023-02-06:
Mind re-opening this?

allangood, 2023-04-06:
I am rewriting the whole project and this will be in the new version!

melyux, 2024-03-22:
Hey Allan, were you able to get this working?

melyux, 2025-01-20:
Maybe you could add it to the current code before the rewrite. Call it a backport

melyux, 2025-05-05:
Can you at least re-open this issue and remove the "not a bug" label?

allangood, 2025-05-05:
Could you help me with this, please? #254

allangood, 2025-05-30:
Done #193