allangood/rtlamr2mqtt #193: Current status of the project

allangood, 2023-04-06:
Hello all,

I currently not happy with the status of the code of this project. It grew too much since my first post. Lots of new features has been added and the add-on, that I particularly don't use, made everything very hard to maintain.

The current code is not clear and hard to change!

I am currently rewriting the whole thing. Yes, from scratch. :)

This means I will not be making any change to the current code, but instead a new code will be pushed very soon.

Thank you for your understanding.

ragde22, 2023-04-19:
with this config:
sleep_for: 0
verbosity: debug
listen_only: true
tickle_rtl_tcp: false
device_id: 0bda:2838

ha_autodiscovery: true
ha_autodiscovery_topic: homeassistant
base_topic: rtlamr
tls_enabled: true

rtltcp: "-s 2048000"
rtlamr: "-unique=true"

I got this in the Log:
[2023-04-18 16:48:14] Using "/data/options.json" config file
[2023-04-18 16:48:14] Fetching default MQTT configuration from http://supervisor/services/mqtt
[2023-04-18 16:48:14] Starting in LISTEN ONLY Mode...
[2023-04-18 16:48:14] Using "/data/options.json" config file
[2023-04-18 16:48:14] Fetching default MQTT configuration from http://supervisor/services/mqtt
[2023-04-18 16:48:14] Configured MQTT sender:
[2023-04-18 16:48:14]  > hostname => core-mosquitto
[2023-04-18 16:48:14]  > port => 1883
[2023-04-18 16:48:14]  > username => addons
[2023-04-18 16:48:14]  > client_id => rtlamr2mqtt
[2023-04-18 16:48:14]  > base_topic => rtlamr
[2023-04-18 16:48:14]  > availability_topic => rtlamr/status
[2023-04-18 16:48:14]  > tls => None
[2023-04-18 16:48:14] Starting rtl_tcp with /usr/bin/rtl_tcp 
[2023-04-18 16:48:16] Starting rtlamr with ['/usr/bin/rtlamr', '-msgtype=all', '-format=json']
[2023-04-18 16:48:16] You should see all utility meters after this line:
[2023-04-18 16:48:16] 16:48:16.166905 decode.go:45: CenterFreq: 912600155
[2023-04-18 16:48:16] 16:48:16.167297 decode.go:46: SampleRate: 2359296
[2023-04-18 16:48:16] 16:48:16.167314 decode.go:47: DataRate: 32768
[2023-04-18 16:48:16] 16:48:16.167325 decode.go:48: ChipLength: 72
[2023-04-18 16:48:16] 16:48:16.167335 decode.go:49: PreambleSymbols: 32
[2023-04-18 16:48:16] 16:48:16.167345 decode.go:50: PreambleLength: 4608
[2023-04-18 16:48:16] 16:48:16.167354 decode.go:51: PacketSymbols: 736
[2023-04-18 16:48:16] 16:48:16.167364 decode.go:52: PacketLength: 105984
[2023-04-18 16:48:16] 16:48:16.167381 decode.go:59: Protocols: r900,scm,scm+,idm
[2023-04-18 16:48:16] 16:48:16.167409 decode.go:60: Preambles: 111110010101001100000,0001011010100011,01010101010101010001011010100011,00000000000000001110010101100100
[2023-04-18 16:48:16] 16:48:16.167434 main.go:124: GainCount: 29
[2023-04-18 16:56:14] 16:56:14.895987 main.go:343: Receiver context cancelled.
[2023-04-18 16:56:14] 16:56:14.896634 main.go:320: read tcp 127.0.0.1:41334->127.0.0.1:1234: i/o timeout
[2023-04-18 16:56:14] io.ReadFull
[2023-04-18 16:56:14] main.(*Receiver).Run.func2
[2023-04-18 16:56:14] 	/go/pkg/mod/github.com/bemasher/rtlamr@v0.9.3/main.go:181
[2023-04-18 16:56:14] runtime.goexit
[2023-04-18 16:56:14] 	/usr/local/go/src/runtime/asm_arm64.s:1172

Looks like it shutdown, what I missing?

allangood, 2023-04-19:
Please, don´t hijack this with a question.
Thanks.

Kashalls, 2023-06-29:
Looking forward to an update. Very interested in running this standalone alongside home assistant, not inside of it.

allangood, 2023-12-31:
A lot of things has happened in my life and the development is very slow at the moment.
A major rewrite is necessary as the project grew way beyond my initial idea.
Lots of features has been added that I don't have/use and it makes hard for me to develop or keep it.
Also, I am not an ADD-ON user, what it makes a lot more difficult to maintain something I don't use on my current setup.

But I am still trying to keep the project alive. Thank you for all your feedback and support :)

allangood, 2025-05-05:
Hello again!
Sorry for the (very) long hiatus, but life happens.

I am getting back to this project and have decided to re-write it from scratch.
The project started as something very simple to meet with my needs, but it got a lot of traction  and became way more popular that I could handle. Lots of features requests and big changes, and I couldn't keep up with all.

With this new version, I am writing the code in a way I can maintain and add features with less pain.

The first commit had the intention to help others with something that was helping me, this hasn't changed! :)

allangood, 2025-05-06:
I am getting close to a new release! #255 
I hope to get a stable release this week.

allangood, 2025-05-26:
> I am getting close to a new release! [#255](https://github.com/allangood/rtlamr2mqtt/pull/255) I hope to get a stable release this week.

I am having a hard time with the add-on and the new format for the sensors.
I will keep this thread updated

allangood, 2025-05-29:
Just merged #260 to the `dev` branch!
Everything seems to be working and it is ready to be tested.

allangood, 2025-06-02:
New stable version released!

closed 2025-06-02 by allangood