allangood/rtlamr2mqtt #207: Allowing for tariffs on meter

arthurbenemann, 2023-05-31:
It would be quite nice to allow for [tariffs](https://www.home-assistant.io/integrations/utility_meter/#tariffs) to be setup as part of a meter, just as an extra configuration similar to "device_class".

Home assistant does not support setting tariff as part of the custumization.yaml, so they have to be added during creation of the utility_meter object. That is why placing the request here. The workaround at the moment is to create clone utility_meters, but that end-ups causing duplication on energy dashboard.

My use case is to split a water meter output based on the state of irrigation system.

allangood, 2025-05-05:
Could you help me with this, please? #254

allangood, 2025-05-30:
This is implemented in the `utility_meter` integration and not in the sensor.

closed 2025-05-30 by allangood