allangood/rtlamr2mqtt #208: Fail to capture data from meter

markwoodward86, 2023-07-06:
Honestly im not sure why this just started, i figured it coincided with the latest homeassistant core update but i rolled back to 2023.6.3 and it persists.

I seem to not see any meter reads any longer. No errors per se but i dont recall seeing this is in the logs before

```
[2023-07-06 18:15:11] 18:15:11.367970 main.go:343: Receiver context cancelled.
[2023-07-06 18:15:11] 18:15:11.368278 main.go:320: read tcp 127.0.0.1:37680->127.0.0.1:1234: i/o timeout
[2023-07-06 18:15:11] io.ReadFull
[2023-07-06 18:15:11] main.(*Receiver).Run.func2
[2023-07-06 18:15:11] 	/go/pkg/mod/github.com/bemasher/rtlamr@v0.9.3/main.go:181
[2023-07-06 18:15:11] runtime.goexit
[2023-07-06 18:15:11] 	/usr/local/go/src/runtime/asm_amd64.s:1594
```

config as follows:

```
general:
  sleep_for: 60
  verbosity: debug
  listen_only: false
  tickle_rtl_tcp: false
  device_id: single
  usb_reset: "009:012"
mqtt:
  ha_autodiscovery: true
  ha_autodiscovery_topic: homeassistant
  base_topic: rtlamr
  tls_enabled: false
  tls_ca: /etc/ssl/certs/ca-certificates.crt
  tls_cert: /etc/ssl/my_self_signed_cert.crt
  tls_keyfile: /etc/ssl/my_self_signed_cert_key.key
  tls_insecure: true
  host: core-mosquitto
  port: 1883
  user: homeassistant
  password: nothingtoseehere
custom_parameters:
  rtltcp: "-s 2048000"
  rtlamr: "-unique=true"
meters:
  - id: 1552820016
    protocol: scm+
    name: production_meter
    format: "#####.##"
    unit_of_measurement: kWh
    icon: mdi:transmission-tower-export
  - id: 1552820019
    protocol: scm+
    name: consumption_meter
    format: "#####.##"
    unit_of_measurement: kWh
    icon: mdi:transmission-tower-import
```
I have reset the adapter physically at the server too. Im running hass os on a proxmox vm. Proxmox is on 7.4 (latest is 8) but this issue suddenly appeared today and proxmox 8 has been out for a while. Could still be hardware... Wanted to add an issue incase you see something obvious.

Thanks for your time and development as always!

justinmbrock, 2023-07-10:
I'm seeing the same problem.

markwoodward86, 2023-07-11:
Actually i wonder if this os related to https://github.com/bemasher/rtlamr/issues/188. Coincidence that there are 2 of us seeing this?

allangood, 2023-07-11:
I/O timeout usually means error accessing the USB device.

closed 2023-07-11 by allangood