allangood/rtlamr2mqtt #219: switch between frequencies for a set time.

FinestDice17, 2023-11-13:
Between my electric, gas and water meters at my home, i need to be able to hop between 912M frequency and 915M frequency. This automatically just sets it to 912 and no matter what i do, it will not hop over to 915. Is there something im missing or can this just not happen with this addon?

tfc87, 2024-01-28:
Looking for the same functionality.  I'm actually wanting to see if I can find my gas and power meters so would like to scan various frequencies.  This worked great for my water meter but would like to run the listen-only mode for other frequencies if possible.

allangood, 2025-05-30:
You can add extra arguments to rtlamr or rtl_tcp in the `custom_parameters` section.
All arguments can be found here: https://github.com/bemasher/rtlamr/wiki/Configuration

The frequency range is from 902MHz to 928MHz, maybe your meter is not supported.

allangood, 2025-05-30:
https://github.com/bemasher/rtlamr/wiki/Compatible-Meters

closed 2025-05-30 by allangood