allangood/rtlamr2mqtt #220: Unable to install from HACS

justin654, 2023-12-06:
I've reset my HA installation on a NUC, however I am unable to install this add on. I get an error message. Everything else has installed without a hitch. Anyone have any suggestions? 

Failed to install add-on
The command '/bin/sh -c go install github.com/bemasher/rtlamr@latest && apt-get update && apt-get install -y libusb-1.0-0-dev build-essential git cmake && git clone git://git.osmocom.org/rtl-sdr.git && cd rtl-sdr && mkdir build && cd build && cmake .. -DDETACH_KERNEL_DRIVER=ON -DENABLE_ZEROCOPY=ON -Wno-dev && make && make install' returned a non-zero code: 128

justin654, 2023-12-08:
@allangood Any chance you have this issue on your radar already?

justin654, 2023-12-08:
> I too have the same issue when trying to install on a RPi 4, but going through the add-on store after adding the repository.

If you are still having this issue, you can give this change a shot #221 it resolved the issue for me.

MikeDembek, 2023-12-08:
I'm seeing this issue as well

justin654, 2023-12-08:
> I'm seeing this issue as well

Try giving the fix I posted above, worked for me. #221

allangood, 2023-12-08:
#221 was merged. Thank you @justin654 for your submission!

MikeDembek, 2023-12-08:
That sorted it for me as well - thanks!

allangood, 2023-12-08:
Awesome! Let's close this issue then!
Thank you for all your reports and feedback!

PR #222 (2023-12-08): Bump version to fix #220
Fix #220
commit b9590efb (2023-12-08): Update config.yaml

Fix #220
commit bb219e5c (2023-12-08): Merge pull request #222 from allangood/allangood-patch-1

Bump version to fix #220
commit 1dcc8ddd (2025-05-06): Update config.yaml

Fix #220
closed 2023-12-08 by allangood