allangood/rtlamr2mqtt #225: No results with -e LISTEN_ONLY=yes -e RTL_MSGTYPE="all"

D34DC3N73R, 2023-12-27:
I just got a Nooelec NESDR Mini USB. My electric meter is a C2S0D, but I am unable to determine the device ID. The text above my FCC info contains characters and numbers, which doesn't seem correct and result in an error. I'm attempting to scan all nearby devices but I get 0 results even after running it for a full 24 hours.

```
$ docker run --rm -ti --privileged -e LISTEN_ONLY=yes -e RTL_MSGTYPE="all" -e TZ="America/Los_Angeles" --device=/dev/bus/usb:/dev/bus/usb allangood/rtlamr2mqtt
[2023-12-26 15:24:26] Starting in LISTEN ONLY Mode...
[2023-12-26 15:24:26] No Supervisor detected.
[2023-12-26 15:24:26] Starting rtl_tcp with /usr/bin/rtl_tcp
[2023-12-26 15:24:28] Starting rtlamr with ['/usr/bin/rtlamr', '-msgtype=all', '-format=json']
[2023-12-26 15:24:28] You should see all utility meters after this line:
[2023-12-26 15:24:28] 15:24:28.671147 decode.go:45: CenterFreq: 912380000
[2023-12-26 15:24:28] 15:24:28.671436 decode.go:46: SampleRate: 2359296
[2023-12-26 15:24:28] 15:24:28.671441 decode.go:47: DataRate: 32768
[2023-12-26 15:24:28] 15:24:28.671444 decode.go:48: ChipLength: 72
[2023-12-26 15:24:28] 15:24:28.671446 decode.go:49: PreambleSymbols: 32
[2023-12-26 15:24:28] 15:24:28.671448 decode.go:50: PreambleLength: 4608
[2023-12-26 15:24:28] 15:24:28.671451 decode.go:51: PacketSymbols: 736
[2023-12-26 15:24:28] 15:24:28.671453 decode.go:52: PacketLength: 105984
[2023-12-26 15:24:28] 15:24:28.671457 decode.go:59: Protocols: scm,scm+,idm,r900
[2023-12-26 15:24:28] 15:24:28.671462 decode.go:60: Preambles: 01010101010101010001011010100011,00000000000000001110010101100100,111110010101001100000,0001011010100011
[2023-12-26 15:24:28] 15:24:28.671469 main.go:124: GainCount: 29
```

I'm new with SDRs in general and the underlying projects this image uses, so if anyone has any suggestions, I'm all ears.

EMiiiiiler, 2024-04-12:
I'm having the same problems too. Similar issues have recommended using a powered USB hub but that ain't working either.

Somehow a week or so ago I managed to get 2 meter readings, but since then nothing

allangood, 2025-05-30:
Some meters are not supported due to encrypted protocol.
I can't add support to any protocol that hasn't been added to the RTLAMR project.