allangood/rtlamr2mqtt #230: How do i know what meters my device is picking up?

JoshuaSeidel, 2024-03-20:
`root@3dprinter:/home/pi# docker run --rm -ti -e LISTEN_ONLY=yes -e RTL_MSGTYPE="all" --device=/dev/bus/usb:/dev/bus/usb allangood/rtlamr2mqtt
[2024-03-20 13:36:34] Starting in LISTEN ONLY Mode...
[2024-03-20 13:36:34] No Supervisor detected.
[2024-03-20 13:36:34] Starting rtl_tcp with /usr/bin/rtl_tcp
[2024-03-20 13:36:36] Starting rtlamr with ['/usr/bin/rtlamr', '-msgtype=all', '-format=json']
[2024-03-20 13:36:36] You should see all utility meters after this line:
[2024-03-20 13:36:36] 13:36:36.154052 decode.go:45: CenterFreq: 912600155
[2024-03-20 13:36:36] 13:36:36.154314 decode.go:46: SampleRate: 2359296
[2024-03-20 13:36:36] 13:36:36.154334 decode.go:47: DataRate: 32768
[2024-03-20 13:36:36] 13:36:36.154346 decode.go:48: ChipLength: 72
[2024-03-20 13:36:36] 13:36:36.154357 decode.go:49: PreambleSymbols: 32
[2024-03-20 13:36:36] 13:36:36.154369 decode.go:50: PreambleLength: 4608
[2024-03-20 13:36:36] 13:36:36.154381 decode.go:51: PacketSymbols: 736
[2024-03-20 13:36:36] 13:36:36.154393 decode.go:52: PacketLength: 105984
[2024-03-20 13:36:36] 13:36:36.154413 decode.go:59: Protocols: idm,r900,scm,scm+
[2024-03-20 13:36:36] 13:36:36.154429 decode.go:60: Preambles: 01010101010101010001011010100011,00000000000000001110010101100100,111110010101001100000,0001011010100011
[2024-03-20 13:36:36] 13:36:36.154471 main.go:124: GainCount: 29
[2024-03-20 13:36:39] {"Time":"2024-03-20T13:36:39.899977259Z","Offset":0,"Length":0,"Type":"SCM","Message":{"ID":30552336,"Type":13,"TamperPhy":0,"TamperEnc":0,"Consumption":1846959,"ChecksumVal":36378}}
[2024-03-20 13:37:14] {"Time":"2024-03-20T13:37:14.033395287Z","Offset":0,"Length":0,"Type":"SCM+","Message":{"FrameSync":5795,"ProtocolID":30,"EndpointType":171,"EndpointID":73060126,"Consumption":3195649,"Tamper":9472,"PacketCRC":54843}}
[2024-03-20 13:37:42] {"Time":"2024-03-20T13:37:42.503846416Z","Offset":0,"Length":0,"Type":"SCM","Message":{"ID":30552336,"Type":13,"TamperPhy":0,"TamperEnc":0,"Consumption":1846959,"ChecksumVal":36378}}
[2024-03-20 13:37:59] {"Time":"2024-03-20T13:37:59.030101268Z","Offset":0,"Length":0,"Type":"SCM+","Message":{"FrameSync":5795,"ProtocolID":30,"EndpointType":171,"EndpointID":73060126,"Consumption":3195649,"Tamper":9472,"PacketCRC":54843}}`

dolfelt, 2024-05-01:
You will want to compare the ID in the messages to the ID/Barcode on your meter. It depends on the meter type, but the ID should be listed somewhere, and then you can filter out messages based on that.

You could also compare your `Type` field with the list [here](https://github.com/bemasher/rtlamr/blob/master/meters.csv). Looks like the `Type: 13` indicates a possible water meter. Hope this helps!

allangood, 2025-05-30:
I will close this as preparation for the next release.

closed 2025-05-30 by allangood