allangood/rtlamr2mqtt #232: Can't install Addon

LordPsycho202, 2024-05-08:
Had some issues with the Addon after the 2024.5.1 upgrade.  Attempted to reinstall the addon, and now it won't install.  

I get the following error

The command '/bin/sh -c go install github.com/bemasher/rtlamr@latest && apt-get update && apt-get install -y libusb-1.0-0-dev build-essential git cmake && git clone git://git.osmocom.org/rtl-sdr.git && cd rtl-sdr && mkdir build && cd build && cmake .. -DDETACH_KERNEL_DRIVER=ON -DENABLE_ZEROCOPY=ON -Wno-dev && make && make install' returned a non-zero code: 128

Happy to provide additional details if needed.

cgoudie, 2024-07-23:
I have a newer install, and was able to add the addin this month.

Elderwolf1700, 2024-07-24:
Oops, guess I should have replied. After changing USB extension cables from the provided one it started working.

allangood, 2025-05-30:
Could you try the next version and see if it works?
Instructions here #193

closed 2025-05-30 by allangood