allangood/rtlamr2mqtt #235: Receiver Context Cancelled. I/O Timeout

lohannasberg, 2024-05-28:
I have been running this docker for ~18 months without issue for my gas meter. About a month ago, it randomly stopped working. For reference, [this](https://www.amazon.com/dp/B011HVUEME) is my SDR, plugged into a 6ft USB 2.0 extension cable.

If I run the container from docker compose, I never get an error message' it just "stalls" and it never receives any data

```
Trying to start RTL_TCP: /usr/bin/rtl_tcp -d 0 -s 2048000
RTL_TCP started with PID 9
RTL_TCP is ready to receive connections!
Trying to start RTLAMR: /usr/bin/rtlamr -msgtype=scm -format=json -filterid={{REDACTED}}-unique=true -symbollength=32
RTLAMR started with PID 14
```

When I run the container in listen only mode, It successfully reads data from neighboring meters, but it errors out within ~5-15 minutes of running with:

```
main.go:343: Receiver context cancelled.
main.go:320: read tcp 127.0.0.1:59170->127.0.0.1:1234: i/o timeout
io.ReadFull
main.(*Receiver).Run.func2
/go/pkg/mod/github.com/bemasher/rtlamr@v0.9.3/main.go:181
runtime.goexit
/usr/local/go/src/runtime/asm_amd64.s:1650
```

I have tried the following without success:

- Rebooting host
- Reverted back to 2.3.3
- Re-pulling docker image
- Different USB port
- Different USB Extension cable (2.0)
- Non-powered USB hub
- Powered USB hub
- Different SDR ([this](https://www.amazon.com/dp/B0CD7558GT/) one)
- Different Antenna
- Confirmed w/ gas company that my meter has not been changed.

No idea what else to try or what the root cause is here. I figured my SDR went bad and bought a new one, but got the exact same error with the new SDR. 

Looking for advice on how to resolve. Any help is greatly apprcaited.

ve6rah, 2024-09-27:
I'm afraid I don't know the solution, but I'm putting a "me too" here to bump it as I'm having the exact same problem. Worked fine for over a year, now won't last a day before I have to do a full system reboot to get back on track.

allangood, 2025-05-30:
This has been fixed in the next release.
Please, check #193 on how to test it.

allangood, 2025-05-30:
I will close this as preparation for the next release.