allangood/rtlamr2mqtt #236: Is having the sample meters in my config blocking device discovery?

cgoudie, 2024-07-23:
New to rtlamr2mqtt. Using HAOS. Having a problem and hoping for help.

Using a Nooelec RTL-SDR v5. debug logging is enabled. Everything seems to start up ok:

```
[2024-07-22 21:01:42] Trying to start RTLAMR: /usr/bin/rtlamr -msgtype=scm+,scm,scm+ -format=json -filterid=7823010,6567984,123456789 -unique=true -symbollength=32
[2024-07-22 21:01:42] RTLAMR started with PID 15
<snip>
[2024-07-22 21:01:42] 21:01:42.450849 decode.go:59: Protocols: scm+,scm
[2024-07-22 21:01:42] 21:01:42.450854 decode.go:60: Preambles: 0001011010100011,111110010101001100000
[2024-07-22 21:01:42] 21:01:42.450864 main.go:124: GainCount: 29

```

After this, I don't see another line in the logs even after waiting a day. I'd be surprised if I'm a radio black hole. I'm wondering if that filterid on the command line is limiting the debug information? I tried to delete the sample meters from the HA Add-in config page, but you can't remove them all or it wont let you save.

 How do I get it to where it outputs the debug information, or is there a problem with my equipment?

Nurgak, 2024-07-30:
If you want listen only mode it's [documented here](https://github.com/allangood/rtlamr2mqtt?tab=readme-ov-file#i-dont-know-my-meters-id-what-can-i-do). Basically, instead of executing docker compose, with a mandatory device filter, you just launch it via:
```sh
docker run --rm -ti -e LISTEN_ONLY=yes -e RTL_MSGTYPE="all" --device=/dev/bus/usb:/dev/bus/usb allangood/rtlamr2mqtt
```
This will not apply any filters and output everything to the console, as far as I understand.

cgoudie, 2024-08-04:
My apologies, I lost track of this issue. To those that come here after me. In the general config section change `listen_only: false` to true.

closed 2024-08-04 by cgoudie

talormanda, 2025-03-01:
> My apologies, I lost track of this issue. To those that come here after me. In the general config section change `listen_only: false` to true.

I was confused by this too. Do we set `verbosity:debug` AND `listen_only: true` ? How do you view what is captured then...? Only in the Log?